OpenCPN/OpenCPN #1353: VS2017 build errors

Hakansv, 2019-04-10:
Today's Git master but these errors appeared first time after commit: Merge pull request #1335 from leamas/mingw
```
19>Done building project "SQLITE_CPP.vcxproj".
19>LINK : warning LNK4044: unrecognized option '/s;/DEBUG'; ignored
19>   Creating library C:/Builds/OCPN/OpenCPN/build/Release/opencpn.lib and object C:/Builds/OCPN/OpenCPN/build/Release/opencpn.exp
19>opencpn.vcxproj -> C:\Builds\OCPN\OpenCPN\build\Release\opencpn.exe
19>Error copying file "RelWithDebInfo/opencpn.pdb" to "RelWithDebInfo/opencpn-5.0.0.pdb".
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: The command "setlocal
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: cmake -E copy RelWithDebInfo/opencpn.pdb RelWithDebInfo/opencpn-5.0.0.pdb
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmEnd
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmErrorLevel
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: exit /b %1
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmDone
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd
19>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :VCEnd" exited with code 1.
19>Done building project "opencpn.vcxproj" -- FAILED.
```

leamas, 2019-04-10:
Hej Håkan...

The root cause here is that the default target is changed from Release to ReleaseWithDebInfo, IMHO for good reasons.  Is it that your build job tries to build the Release configuration? Can you build ReleaseWithDebInfo instead?

Hakansv, 2019-04-10:
Hej leamas..
Thanks for the info. Yes it worked fine in ReleaseWithDebInfo instead. And after that is the release build working wo errors as well.
Silly questions.....:
- I suppose the release must be built since that binary is still the one used for the package?
- And I also suppose the release/opencpn.lib is still the one to be used as target lib for the plugins?
- So now I've to build first for ReleaseWithDebInfo  and then for release on every source update?
- Who and what are using \ReleaseWithDebInfo\binaries? (.exe .lib)

Thanks
Håkan
BTW: Still a new warning:
Done building project "TEXCMP.vcxproj".
20>LINK : warning LNK4044: unrecognized option '/s;/DEBUG'; ignored

leamas, 2019-04-10:
> suppose the release must be built since that binary is still the one used for the package?

If I  understand your question correctly, no. 

The background is that prior to latest patches the Release target produced a binary with debug info. Of course, the same was (and is) true for RelWithDebInfo. The net result was that while the Release target did it's job as a default target, there was no way (short of patching the build system) to create a binary *without* debug info.

After the patches things are what they are supposed to be: Release produces binaries and libs without  and RelWithDebInfo produces artifacts with debug info.

Still, the default targets we use are the ones with debug info. This is true for developers for obvious reasons, but also for packagers. Packaging toolchains  "always" build binaries with debug info; in a later step they create separate debug packages and then strips said info.

TL;DR The idea is that from now on we use the RelWithDebInfo artifacts instead of the Release one in all default builds.

leamas, 2019-04-10:
> 20>LINK : warning LNK4044: unrecognized option '/s;/DEBUG'; ignored

Yes, good catch.   I'll try to fix that (if not already fixed) in #1298

leamas, 2019-04-10:
Hm... I just skimmed through the logs in the appveyor build for #1298,  It seems that this is already fixed there.

Hakansv, 2019-04-12:
Builds clean now. Thanks

closed 2019-04-12 by Hakansv