OpenCPN/OpenCPN #1355: Master build broken on Arch Linux - wxWebView and cmake

taruti, 2019-04-11:
Yesterday things worked, today things are broken.

Related to #1329 but caused by new cmake changes? @leamas do you have any idea?

Compiling chart1.cpp fails with:
```
.../OpenCPN/include/AboutFrame.h:64:3: error: ‘wxWebView’ does not name a type; did you mean ‘wxListView’?
```

Hakansv, 2019-04-11:
@taruti If of any help I don't know but it builds in VS2017

[image]

leamas, 2019-04-11:
Well, none of my commits was merged yesterday, I'm innocent ;)

Seems like the OCPN_USE_WEBVIEW compilation constant is set despite that there is no webview found. I have had similar problems. The related output is from CMakeLists.txt around line 565 (webview detection) and the compilation command which most like contains -DOCPN_USE_WEBVIEW

Perhaps could pin down the failing commit with a git bisect?

leamas, 2019-04-11:
Seems like the fix [1f052c0ee11286ef80b5c655548b7e3c33e9da7f] (two weeks ago) have been  reverted by some malicious dev ... which turns out to be me : [59425e56b9] 

This is a rebase error, and we need to revert it asap i. e., use wxWidgets_FIND_COMPONENTS instead of wxWidgets_USE_LIBS.

leamas, 2019-04-12:
At last... this was actually a mess. 

@taruti: Could you test the patch in #1360?

closed 2019-04-12 by nohal

taruti, 2019-04-15:
Master now works again for me, thanks!

PR #1360 (2019-04-12): Restore 1f052c0 cmake wxWidgets detection (#1355).
This is actually a can of worms opened. Part of the problem is the
use of wxWidgets_USE_LIBS instead of wxWidgets_FIND_COMPONENTS which
does not work on modern cmakes, fixed in 1f052c0 but lost when
rebasing a later PR.

Another thing is the C code using #ifdef OCPN_USE_WEBVIEW. This flag
is not always correct, webview could be disabled due to e. g.,
missing backends even if configured. In this case wxUSE_WEBVIEW is
set to 0, otherwise 1. So the correct test is #if wxUSE_WEBVIEW,
augmented with a configuration HAVE_WEBVIEW flag.

Current code  [... cut]