OpenCPN/OpenCPN #1358: Multiple Chart Display Keyboard Zoom & Auto-Follow Issue

davidsanner, 2019-04-11:
When two chart views are being displayed and after clicking/selecting one chart, the keyboard zoom-in (+/=) effects one  of the two charts while the zoom-out (-) effects the other chart. 

Also when the auto-follow option is enabled in 5.0, moving the chart doesn't disable auto-follow like it did in 4.x  so the display keeps re-centering / fighting the user input (unless vessel is moved fully off the screen & then auto-follow is disabled).

nohal, 2019-04-11:
> When two chart views are being displayed and after clicking/selecting one chart, the keyboard zoom-in (+/=) effects one of the two charts while the zoom-out (-) effects the other chart.
> 

Can't reproduce, what is your OS? By "chart" you refer to chart canvas (or chart view in your first sentence), right, right?

> Also when the auto-follow option is enabled in 5.0, moving the chart doesn't disable auto-follow like it did in 4.x so the display keeps re-centering / fighting the user input (unless vessel is moved fully off the screen & then auto-follow is disabled).

This is a long requested feature.

davidsanner, 2019-04-11:
Testing on a fully patched RaspberryPi3 running raspbian/debian 9.8 w/ 4.14.98-v7+ kernel.
By two chart view I mean when the new duel charts are being display.  Duel chart canvas layout.

--> However, I just tested (over VNC this time) and I was not able to reproduce.  Odd.

At the time when this occurred I had previously done some fine zooming in/out but no luck recreating the input behavior now.   

I guess it's ok to close this out and if/when I can reproduce reliably I'll open another issue w/ more details.  Thanks.

closed 2019-04-11 by nohal