OpenCPN/OpenCPN #1364: new cmake.txt remove symbol on debug build

did-g, 2019-04-15:


leamas, 2019-04-15:
Oops... platform?

did-g, 2019-04-15:
linux
diff --git a/CMakeLists.txt b/CMakeLists.txt
index eef27848b..f0bcab406 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -615,7 +615,7 @@ MESSAGE (STATUS "*** Will install to
${CMAKE_INSTALL_PREFIX}  ***")
   #  if(CMAKE_COMPILER_IS_GNUCXX)
   #      set(CMAKE_CXX_FLAGS "-O2")        ## Optimize
   #  endif()
-  set(CMAKE_EXE_LINKER_FLAGS "-s")  ## Strip binary
+#  set(CMAKE_EXE_LINKER_FLAGS "-s")  ## Strip binary

 IF(QT_ANDROID)
   INCLUDE_DIRECTORIES("${Qt_Base}/android_armv7/include/QtCore")

nohal, 2019-04-15:
This does not exist anymore...

did-g, 2019-04-15:
Great.

What's wrong with verbose by default and make -s ?
Now with - Silent build mode. OK
you can't get make command output without zillion of junk, ie with the 100%
undocumented hack make VERBOSE=1

leamas, 2019-04-15:
The VERBOSE=1 hacks needs at least a note in CMakeLists.txt, agreed. Will make a PR.

What's worse is that we still are hardcoding a lot of cmake standard variables like CMAKE_CXX_FLAGS making it impossible for users to add options. Need to be fixed...

nohal, 2019-04-15:
I definitely expect `CMAKE_VERBOSE_MAKEFILE` to be `ON` unless `OCPN_VERBOSE` is switched to `OFF` explicitly.

did-g, 2019-04-15:
Hi,
Not opening new issue but it could be one or a follow up of a closed one:
no CMakeCache.txt

cmake  -DCMAKE_BUILD_TYPE=Debug ..

config file generated:
#ifndef OCPN_CONFIG_H
#define OCPN_CONFIG_H

/* #undef LINUX_CRASHRPT */
/* #undef OCPN_USE_CRASHRPT */
/* #undef OCPN_HAVE_X11 */

// Garmin Host Mode support
#define USE_GARMINHOST

#define OCPN_USE_LZMA

#define OCPN_USE_CURL

#define USE_S57

/* #undef USE_LIBELF */

#define USE_GARMINHOST

/* #undef ocpnUSE_NEWSERIAL */

// Enable dark mode on modern MacOS.
/* #undef OCPN_USE_DARKMODE */

/* #undef ocpnUSE_GLES */

#ifndef TIXML_USE_STL
#define TIXML_USE_STL
#endif

#define OPENGL_FOUND
#define HAVE_PORTAUDIO
#define HAVE_SYSTEM_CMD_SOUND
#define HAVE_SNDFILE

I don't have LIBELF so it's ok but at least ocpnUSE_NEWSERIAL seems wrong,
not sure about other undef, on linux there's X11, isOCPN_HAVE_X11  still
used?

leamas, 2019-04-15:
Too late now, but I will look into this tomorrow.  

BTW: I actually think it would be easier to handle this in separate issues, here are several unrelated things.

leamas, 2019-04-15:
@nohal: Before I go to bed:
> I definitely expect CMAKE_VERBOSE_MAKEFILE to be ON [by default].

Are you sure?

One reason that ocpn has had so many compiler warnings is probably that they haven't been visible in the noisy, verbose output from standard builds. For that reason I think silent builds are the best for casual builds since they make the compiler output much more visible.

OTOH, I agree when it comes to packaging and CI builds where you typically need all info you can get in 
the logs. But this is easy to accomplish in build scripts, adding make VERBOSE=1 or so.

@did-g: The ocpnUSER_NEWSERIAL is a plain bug. Good catch!

nohal, 2019-04-15:
The only reason OpenCPN or anything else has so many compiler warnings is that nobody cares about fixing them, IMO.
And yes, I'm 100% positive I prefer the verbose output from any build, whoever does not like it may use `make -s` or configure with verbose output disabled.

leamas, 2019-04-16:
> The only reason OpenCPN or anything else has so many compiler warnings is that nobody cares about fixing them, IMO.

Indeed. But this the chicken and egg problem: If you have a build with a lot of warnings and you make a change which gives yet another the impression is that since nobody cares, why should I?

IMHO we need to come to point where it at least isn't OK to add new warnings. We should of course also handle the remaining ones we have now when the list is more manageable.

nohal, 2019-04-16:
I think we have exhausted the scope of this ticket by now

closed 2019-04-16 by nohal