OpenCPN/OpenCPN #1367: Windows package. Program version label is missing in other language than English.

Hakansv, 2019-04-16:
Since some days? is the packaging set-up missing the version label when I use Swedish as the installation language. 
See the sequence below. First English then Swedish.
(Version 5.0.2 is my own way to distinguish between my packages. Not a master bug:))

[image]

[image]
[image]
[image]

leamas, 2019-04-16:
ouch... this looks complicated. First check: the buildwin/ directory hasn't been changed sinc 2018.

As for NSIS.template.in it is now generated from NSIS.template.in.in. The only diff between these files is two added comment lines.

    ; Enabled on MINGW builds:   
    ;!define NSIS_PACKEDVERSION true

and on Windows the .in.in file is copied verbatim to NSIS.template.in.

TL;DR If neither buildwin nor NSIS.template.in has changed, what could this be about?

leamas, 2019-04-17:
Some hunting, making notes to not forget. The main NSIS.template.in.in contains:

    #CPACK_NSIS_DISPLAY_NAME "${CPACK_PACKAGE_NAME} Version ${PACKAGE_VERSION}"

which seems to indicate the default value for CPACK_NSIS_DISPLAY_NAME. This *could* be the failing string in this bug.... The Swedish language file contains:

    ;added on 2011-06-13
    #LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
    LangString ^NameDA ${LANG_SWEDISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"

which looks perfectly sane. I presume that neither cpack/cmake nor NSIS versions have changed in the last few days when the bug appeared.

@Hakansv: perhaps you could try to uncomment the line in NSIS.template.in.in above (remove the hashmark) and add some garbage besides Version just to see if this is indeed the string involved?

leamas, 2019-04-17:
And... could you possibly try this patch: 

    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index 410f2eea4..cc700374b 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -2456,7 +2456,7 @@ if (WIN32 AND NOT UNIX)
 
       # These lines set the name of the Windows Start Menu shortcut and the icon
       # that goes with it
    -  set(CPACK_NSIS_DISPLAY_NAME "${CPACK_PACKAGE_NAME} ${PACKAGE_VERSION}")
    +  set(CPACK_NSIS_DISPLAY_NAME "${CPACK_PACKAGE_NAME} Version ${PACKAGE_VERSION}")
       set(CPACK_PACKAGE_ICON "${CMAKE_SOURCE_DIR}/src/bitmaps\\\\opencpn.ico")
 
       set(CPACK_PACKAGE_FILE_NAME "${PACKAGE_NAME}_${PACKAGE_VERSION}_setup")

Hakansv, 2019-04-17:
@leamas 
First: the Windows Start Menu shortcut and the icon are correct so that's not the problem?
The version label is missed only inside NSIS installer dialog windows. Program files and everything else are as they should be.

Your suggested patch to CMakeLists.txt yield no difference. 

NSIS.template.in.in or before NSIS.template.in have no difference here
;added on 2011-06-13
	LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
    LangString Installer ${LANG_ENGLISH} "Installer"

But I performed your test to:
```
;added on 2011-06-13
	#LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
	LangString ^NameDA ${LANG_SWEDISH} "@CPACK_PACKAGE_NAME@ Version x @CPACK_PACKAGE_VERSION@"
```
And then the output was as the shot below.  - Strange, why is that needed now? And the English installer is still working correct although commented out. 
Spanish, French or any other tried language are not working.

[image]

[image]

leamas, 2019-04-18:
Ws should need someone who actually knows this NSIS stuff, I'm more or less lost. 

However, if I get it right, the translations works as gettext. That is, the original text is in English, untranslated. The language files contains translations of the English strings.  When translating, the English is presented as a comment to the translator. So the English, untranslated version works but not the translated ones.

And we have some progress: It's about this very string. I assume that the rest of the translation  looks OK, right?

Hakansv, 2019-04-18:
All translations are correct. It's not about that. I think we've changed the way ${PACKAGE_VERSION} or its components are constructed so somewhere in the NSIS chain, when it switches from English to any other language the version is lost?
Håkan

leamas, 2019-04-18:
> I  think we've changed the way ${PACKAGE_VERSION} or its components are constructed so 

I have had the same idea. Should be easy to check by adding 

    message(STATUS "PACKAGE_VERSION: ${PACKAGE_VERSION}")

to CMakeLists.txt around line 2590, just before `Include(CPack)`. And then check the cmake output after removing CMakeCache.txt.

leamas, 2019-04-18:
Bummer, I could do that myself usinfg  appveyor. Doing so, I get

    PACKAGE_VERSION: 5.0.0+0abb40c

which *might* be the culprit -- the last +0bb40c are new. Need to make a patch to test, but not now...

leamas, 2019-04-19:
hm... looking into this it looks like the *+0abb40c* suffix is only generated if we feed` -DCI_BUILD=ON` to cmake -- and the appveyor build has indeed this build flag. Do you use it? If so, what happens if you omit  it?

rgleason, 2019-04-19:
I believe this might have been done to distinguish appveyor Opencpn v5.0 builds, but you probably know that?

leamas, 2019-04-19:
Seems so. That said, if you rebuild without` -DCI_BUILD=ON,` is the bug still there?

Hakansv, 2019-04-22:
Social activities for some days......
The "-DCI_BUILD=ON" I don't quite understand. I don't use any switches for cmake other than "cmake -T v141_xp .."
But what I've found so far is that this code:
```
;added on 2011-06-13
#LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
LangString ^NameDA ${LANG_SWEDISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
```
In Langstrings_SWEDISH.nsh.in is found all the way through CMakeList.txt -> NSIS.template.in -> NSIS.template.in.in. No problem. (If add text after "Version" it's printed in the installer message dialog.) But the info in: @CPACK_PACKAGE_NAME@ and @CPACK_PACKAGE_VERSION@ are not found from "Langstrings_SWEDISH.nsh.in". But how to trace that back I don't know?
The same English parts in NSIS.template.in.in are working though?
Håkan

Hakansv, 2019-04-22:
I also tried to bisect down in master to find the causing commit but there' to many build errors to handle to make it workable.

leamas, 2019-04-22:
>  I don't use any switches for cmake other than "cmake -T v141_xp .."

This means that the version string is 5.0.0, not 5.0.0+deadbee -- the latter form is used when -DCI_BUILD=ON.  From this is seems that ${PACKAGE_VERSION} isn't the culprit.

> (If add text after "Version" it's printed in the installer message dialog.) But the info in: @CPACK_PACKAGE_NAME@ and @CPACK_PACKAGE_VERSION@ are not found from "Langstrings_SWEDISH.nsh.in"

But it's not only that... if you add some text after "version",  @CPACK_PACKAGE_NAME@ and @CPACK_PACKAGE_VERSION@  are actually found, as of https://github.com/OpenCPN/OpenCPN/issues/1367#issuecomment-484256479

I'm lost.  Just to make sure: you have tested complete rebuilds from scratch, right?

leamas, 2019-04-22:
hm... It's about this line:

     #CPACK_NSIS_DISPLAY_NAME "${CPACK_PACKAGE_NAME} Version ${PACKAGE_VERSION}"

When you tested it you uncommented  and added some garbage  after *Version*. What happens if you just uncomment it, with the string otherwise as-is?

Hakansv, 2019-04-22:
What I tested in Langstrings_SWEDISH.nsh.in was like this:
```
;added on 2011-06-13
#LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"
LangString ^NameDA ${LANG_SWEDISH} "@CPACK_PACKAGE_NAME@ Version Garbage @CPACK_PACKAGE_VERSION@"
```
The commented  "#LangString ^NameDA ${LANG_ENGLISH..........."  I see as an information to the translator only? The Swedish LangString, if needed, would be called from this string in NSIS.template.in.in:
`LangString ^NameDA ${LANG_ENGLISH} "@CPACK_PACKAGE_NAME@ Version @CPACK_PACKAGE_VERSION@"`

And BTW: My repository is rebased and built from scratch. And the appveyor built from Github has the same bug.
I've though some missing files in the cmake build but I suppose not related? See attached file.
Håkan

[cmake_output.txt](https://github.com/OpenCPN/OpenCPN/files/3103363/cmake_output.txt)

closed 2019-04-22 by nohal

Hakansv, 2019-04-22:
@nohal 
Thanks! Works.

commit fc6370a1 (2019-04-22): Configure Win installer i18n after the metadata

Fixes #1367

The localization depends on CPack variables so we must do it
after setting the CPACK_* variables.