OpenCPN/OpenCPN #1369: Win10. O crash at start up.

Hakansv, 2019-04-16:
- O crashes in release mode at program start up. Code from today's master. See log extract below.
- Running in debug mode yields no crash.
- Code to and including this commit works wo crash.
a4429e640
Merge pull request #1360 from leamas/master-1355

From log file at crash: 
(Note the folder "OpenCPN\s57data" is not installed by the present install package. A package built from before the mentioned commit above does install the \s57data\)
Translation: fel 3: Det går inte att hitta sökvägen = Error 3. Can't find the path
```
16:44:00:    Error: Could not load S57 ClassInfo from C:\Program Files (x86)\OpenCPN\s57data
16:44:00: Fel: kan inte öppna fil "C:\Program Files (x86)\OpenCPN\s57data\s57attributes.csv" (fel 3: Det går inte att hitta sökvägen.)
16:44:00:    Error: Could not load S57 Attribute Info from C:\Program Files (x86)\OpenCPN\s57data
16:44:00: Fel: kan inte öppna fil "C:\Program Files (x86)\OpenCPN\s57data\s57objectclasses.csv" (fel 3: Det går inte att hitta sökvägen.)
16:44:00:    Error: Could not load S57 Feature Info from C:\Program Files (x86)\OpenCPN\s57data
16:44:00: ChartSymbols ConfigFile not found: C:\Program Files (x86)\OpenCPN\s57data\chartsymbols.xml
16:44:00:    S52PLIB: Cannot open S52 rules file: C:\Program Files (x86)\OpenCPN\s57data/
16:44:00:    S52PLIB Initialization failed, oesenc_pi disabling Vector charts

Every crash ends here:
16:44:01: WMM: WMM model data loaded from file C:\Program Files (x86)\OpenCPN\plugins\wmm_pi\data\WMM.COF.
16:44:01: oesenc_pi:  Calculated pix/mm = 4.42071
16:44:01: oeSENC_PI:  OpenGL-> Minimum cartographic line width:  1.0

```

leamas, 2019-04-16:
hm... Is this to say that the code as of 2139eca5f is OK?   

If so, do you know at which commit it actually breaks?

leamas, 2019-04-16:
hm... (again): Could you try the attached past? I *think* it should make the trick...

[0001-build-bugfix-for-10bbf71f1.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3085693/0001-build-bugfix-for-10bbf71f1.patch.gz)

nohal, 2019-04-16:
Checking the latest CI product, there also are `include` and `lib` dirs with content from SERIAL erroneously included in the Windoze installer now.

nohal, 2019-04-16:
And `share/doc/opencpn` content also seems weird (`changelog`, `copyright`, `serial_example.cc`)

leamas, 2019-04-16:
> Checking the latest CI product, there also are include and lib dirs with content from SERIAL erroneously included in the Windoze installer now.

The culprit is most likely libs/serial/CMakeLists.txt... it seems to install without checking platform. Could you possibly have a look?

leamas, 2019-04-16:
> And share/doc/opencpn content also seems weird (changelog, copyright, serial_example.cc)

This is actually exactly as intended -- this is the standard packaging docs. They are probably not that useful on non-linux platforms, but I have presumed they are no harm in those cases.

nohal, 2019-04-16:
Why `serial_example.cc`?

leamas, 2019-04-16:
because it ... docs, that's how example code is handled.

nohal, 2019-04-16:
Until now I though we create and distribute a desktop application with privately statically linked SERIAL library, so why do we distribute this file exactly?

leamas, 2019-04-16:
> The culprit is most likely libs/serial/CMakeLists.txt... it seems to install without checking platform. Could you possibly have a look?

More exactly this snippet:

    install(TARGETS ${PROJECT_NAME}
        ARCHIVE DESTINATION "${CMAKE_INSTALL_LIBDIR}"
        PUBLIC_HEADER DESTINATION "${CMAKE_INSTALL_INCLUDEDIR}"
    )

Which seems fine on linux and perhaps also osx, but not windows. Question becomes what to do with respect to platform here...

nohal, 2019-04-16:
Before I start looking, why do we install a library and headers for something we link (or at least should link) statically and is not our product on any platform?

leamas, 2019-04-16:
To be honest, I have just used the upstream CMakeLists here, that's why. And it's certainly not a good reason... perhaps we should just drop it?

nohal, 2019-04-16:
OK, so it seems that we both agree on
```diff
diff --git a/libs/serial/CMakeLists.txt b/libs/serial/CMakeLists.txt
index e21437e1b..3b8b6a8d8 100644
--- a/libs/serial/CMakeLists.txt
+++ b/libs/serial/CMakeLists.txt
@@ -94,13 +94,6 @@ else()
     target_link_libraries(${PROJECT_NAME} setupapi)
 endif()

-install(TARGETS ${PROJECT_NAME}
-    ARCHIVE DESTINATION "${CMAKE_INSTALL_LIBDIR}"
-    PUBLIC_HEADER DESTINATION "${CMAKE_INSTALL_INCLUDEDIR}"
-)
-install(FILES examples/serial_example.cc DESTINATION ${CMAKE_INSTALL_DOCDIR})
-
-
 # Other targets: test program, pkg-config and tests.
 if (ENABLE_TEST_PROGRAM)
     add_executable(serial_example examples/serial_example.cc)
```
right?

leamas, 2019-04-16:
Right.

leamas, 2019-04-16:
With this sorted out let's return focus to Håkan's blocking error. I really think the patch (https://github.com/OpenCPN/OpenCPN/issues/1369#issuecomment-483716100) should make it, but it needs some testing.

nohal, 2019-04-16:
If we want to get rid of `USE_S57`, we should probably do it well, as in also everywhere in the code, I think. So I would probably prefer fixing `OCPN_USE_S57` for now.

leamas, 2019-04-16:
> f we want to get rid of USE_S57, we should probably do it well, as in also everywhere in the code, I think. 

Yes, I was a lazy human being here, agreed.

> So I would probably prefer fixing OCPN_USE_S57 for now.

Anything is possible but it's broken already in 4.8.6 (possibly earlier) and I don't see the point trying to fix an obviously unused feature. 

Possible strategy: merge the fix above to make master build, short -term. And the go after all USE_S57 in the code -- it's not hard, just painful.

nohal, 2019-04-16:
I think this is a better temporary solution as it will keep it clear the stuff is still hanging around

closed 2019-04-16 by nohal

leamas, 2019-04-16:
OK, now I see what you mean. Makes sense. Thought you were talking about actually fixing the option so that it could be set to OFF. Will make a complete PR later, time permitting.

nohal, 2019-04-16:
Well, that would be even better... I would wait for @bdbcat before investing effort into complete removal of `USE_S57` from the codebase.

Hakansv, 2019-04-16:
Thanks you boys. Works!  👍

bdbcat, 2019-04-17:
leamas..
OK to kill USE_S57 forever.  Legacy from years ago, attempting to reduce binary app size.  No longer a priority ;)
Dave

commit e1239f29 (2019-04-16): Add missing OCPN_USE_S57 option (Closes #1369)