OpenCPN/OpenCPN #1370: Current master compile leads to incompatible plugins

Carstijn, 2019-04-16:
Hi, I tried to build OpenCPN because there are no packages available for Fedora 29.

With the current master (8411a23c3aecef7ee8f77a80e69da3fd787397a0) OpenCPN compiles just fine but, when I start the application the build-in plugins are incompatible.

Does anyone have a hint what could go wrong here?

cmake command used:
cmake -DCMAKE_INSTALL_PREFIX=/usr ../

opencpn.log snippet:
20:10:25 CEST: PlugInManager searching for PlugIns in location /usr/local/lib64/opencpn
20:10:25 CEST:     Incompatible plugin detected: /usr/local/lib64/opencpn/libgrib_pi.so
20:10:27 CEST:     Incompatible plugin detected: /usr/local/lib64/opencpn/libdashboard_pi.so
20:10:28 CEST:     Incompatible plugin detected: /usr/local/lib64/opencpn/libchartdldr_pi.so
20:10:29 CEST:     Incompatible plugin detected: /usr/local/lib64/opencpn/libwmm_pi.so
20:10:31 CEST:     Incompatible plugin detected: /usr/local/lib64/opencpn/liboesenc_pi.so

[cmake full output.txt](https://github.com/OpenCPN/OpenCPN/files/3086356/cmake.full.output.txt)

nohal, 2019-04-16:
Either the installation logic is broken or you are loading a completely different set of plugins.
With `CMAKE_INSTALL_PREFIX=/usr` the plugins can never end in `/usr/local`

Carstijn, 2019-04-16:
@nohal  you are correct, the plugin location is actually a simlink from /usr/local/lib64/opencpn to /usr/lib64/opencpn

nohal, 2019-04-16:
Then something else is broken (No wonder with what we do to the build process right now). I have enough fun with Windows right now, will get back to this later.

Carstijn, 2019-04-16:
@nohal in case you are reworking this part, I also tested with the v5.0.x branch. This gives the same result.

nohal, 2019-04-16:
There are no changes at all in the `v5.0.x` branch, same stuff for years...

leamas, 2019-04-16:
Could be related to the new plugin load paths code. Basically by default plugins are loaded from the three locations` ~/.local/lib:/usr/local/lib:/usr/lib` corresponding to user, site and application. You could tweak this using environment variables, see the new manpage opencpn.1

nohal, 2019-04-16:
For me this is OK on amd64 Linux so the best guess still is you have a mismatch between the executable and the plugin SOs and one of them is not what you think. Can you try to get rid of the plugin libraries and do a new build/install to make sure they are correct?

Carstijn, 2019-04-16:
I have just cleaned my system of old files and that did not help. An other test, I tried to run from the build directory:
GDK_BACKEND=x11 OPENCPN_PLUGIN_DIRS=./plugins/dashboard_pi/ ./opencpn
this still gives the incompatible error

nohal, 2019-04-16:
What system is that exactly?
What do you get from `ldd /usr/local/lib64/opencpn/libdashboard_pi.so`?

Carstijn, 2019-04-17:
Its a 64bit Fedora 29 system. Here is the output of ldd for both my build directory and the /usr/lib64 location.
[ldd_output_usr_lib64.txt](https://github.com/OpenCPN/OpenCPN/files/3088200/ldd_output_usr_lib64.txt)
[ldd_output_build.txt](https://github.com/OpenCPN/OpenCPN/files/3088201/ldd_output_build.txt)

leamas, 2019-04-17:
I might have found it. Could you please submit your config.h (probably in build/include/config.h)

Carstijn, 2019-04-17:
@leamas here is my config.h
[config.h.txt](https://github.com/OpenCPN/OpenCPN/files/3089840/config.h.txt)

leamas, 2019-04-17:
As I suspected, it's about USE_LIBELF. Thanks for feedback!

You can try the attached patch, it should resolve this. Otherwise I think we will be able to rectify master Real Soon (tm).

[0001-build-Respect-USE_LIBELF-configuration-1370.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3090091/0001-build-Respect-USE_LIBELF-configuration-1370.patch.gz)

PR #1375 (2019-04-17): build: Respect USE_LIBELF configuration (#1370)
Bug: The results from the  CMakeLists code looking for libelf was
not visible in the cpp code. The libelf section was also misplaced
in CMakeLists.

Closes: #1370
commit 65ccc18e (2019-04-17): build: Respect USE_LIBELF configuration (#1370)

Bug: The results from the  CMakeLists code looking for libelf was
not visible in the cpp code. The libelf section was also misplaced
in CMakeLists.

Closes: #1370
closed 2019-04-17 by nohal