OpenCPN/OpenCPN #1383: Raspberry Pi cannot cómplice WMM plugin

transmitterdan, 2019-04-20:
The latest git master crashes my Raspberry Pi GCC compiler when compiling the wmm plugin. I can’t tell what is offending the compiler. Has anyone compiled from master with a freshly updated Pi?

nohal, 2019-04-20:
The crash is what exactly? Do you have memory (=What happens if you create a swapfile to make sure you do)?

transmitterdan, 2019-04-22:
No specific cause us mentioned. It just says to report it as a bug to the GCC developers.

I found a workaround. If I compile wmm_pi.cpp manually then run make it continues on. So my plan is to clear out the build folder and rebuild from scratch.

I watch memory during the build process. There are some source files that require lots of memory such that make -j 2 slows to a crawl when swapping starts. A plain make with only 1 thread does not run out of memory AFAICT.

transmitterdan, 2019-04-23:
Deleted the build folder and reran cmake.  Pulled latest git master and it builds now without issues.

closed 2019-04-23 by transmitterdan