OpenCPN/OpenCPN #1384: Grib V 5.0.0 truncates pressure label on OpenSUSE.

dominig, 2019-04-23:
I have notice that the pressure label data (in hPa) is truncated with the version 5.0.0 of OpenCPN under Linux OpenSUSE.
See attached image. Rge leading "9" is missing.
[image]

rgleason, 2019-04-23:
Not critical, but it would also be nice to have "L" and "H"  labeled.

did-g, 2019-04-23:
That's the proverbial not a bug but a feature, right click on Pressure:
[image]

Have to update doc

rgleason, 2019-04-23:
That's new to me. Thanks.  Will add it to the bottom of Wxfax wiki.

nohal, 2019-04-23:
> Will add it to the bottom of Wxfax wiki.

But we are talking about the grib plugin here, not weatherfax.

rgleason, 2019-04-23:
Sorry, read too fast. Will add to grib_pi wiki. Thanks.

rgleason, 2019-04-23:
See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:included_plugins:grib_weather#pressure_isobar_numbers_-_short_long

closed 2023-04-29 by nohal