OpenCPN/OpenCPN #1386: Chart Toolbar Misplaced

robmadden, 2019-04-23:
OpenCPN 5.0.0

The chart toolbar in my installation is now bound to the upper left corner of my screen. I'm unsure when this first started happening but I believe it may have been after tinkering with the layout for the first time. I will post more information as I go along.

I am seeing a possibly related error in the console:

```
(opencpn:2786): Gtk-CRITICAL **: IA__gtk_widget_set_size_request: assertion 'width >= -1' failed
```

[image]

nohal, 2019-04-23:
That GTK message sure has nothing to do with it.
What display manager is it?

robmadden, 2019-04-23:
LXDM on Raspbian.

Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

robmadden, 2019-04-23:
It appears the chart panel toolbar is not bound to the application window whatsoever.
 
[image]

juanman2, 2019-06-18:
I can see something similar happening when I use CMD-TAB, on OSX to flip through windows.  The chart panel remains on page hovering over the new page, in this case, Chrome displaying a Diesel parts web page. 

[image]

I can try and fix what I see, and hopefully it will fix the other problem as well, even though I don't know how to reproduce that?  If yes, any hints as to where I should look?

nohal, 2019-06-18:
@juanman2 this is a different issue - on macOS wxWidgets library sucks badly on managing the z-order of the windows. You will find quite a bit of hacky code in OpenCPN trying to deal with it without giving up on using the TLWs for the toolbars with variable success and you will probably also find out that it is not as simple to fix as you thought when you first saw it...

bdbcat, 2021-04-28:
Closing, inactive

closed 2021-04-28 by bdbcat