OpenCPN/OpenCPN #1393: RFE: Optionally use chart data for routes

taruti, 2019-05-02:
I'm planning to add support for using chart data (e.g. RECTRC channels) optionally when creating routes. In narrow archipelagos making routes that follow marked channels is useful and creating those manually is not fun.

I am envisioning that when this mode is turned on adding a route point, when the previous and new route points are (near) on a channel, creates a route that follows the channel. 

Open questions at this point:
1) Use S57 data from near current position from the loaded chart or go through the data first and save it in a suitable secondary format.

2) Can this be done as a plugin or should it be in the main application?

3) Any hints on code to study that walks non-trivial S57 objects in OpenCPN/Plugins? Currently I am looking at the "Object Query" implementation?

did-g, 2019-05-04:
Hi,
Not exactly the same  but I did a POC last year for extracting objects from S57 and using them in routing.
Search for 's57 hazards' patches in 
https://github.com/did-g/OpenCPN.git

bcn2, 2019-05-04:
What about the encrypted vector formats like S-63 and oeSENC?
Links as well to the question of a "route-checker", in the context of what did-g is mentioning.
What about to query the plug-in with a vector - the routes are consisting from vectors - and getting returned defined hazards?
(the interface or API would need to prevent from "harvesting" all objects of the protected content)

taruti, 2019-05-06:
Thanks for the link, looking at those patches!

@bcn2 for encrypted formats there already is an API to get stuff (`GetPlugInObjRuleListAtLatLon`) for ObjectQuery and did-g's hazard patches add an another API (`GetHazards`).

Of course if protecting the charts and supporting plugins is hard, then a separate database of recommended tracks could be used. But keeping that up-to-date is more pain for users.

did-g, 2019-05-06:
On the other hand someone able to use this future API is already able to figure out how to extract the unprotected full chart ...

S63 encryption scheme is 100% broken. If you get your hand on an encrypted chart brute force cracking is cheaper than buying it. Key is very small and first bytes are known, it's a zip file. Today 48 bits key is not encryption it's obfuscation.

All this stuff is a big waste of time and money, for recreational boating lobby for free, unprotected charts, I do.

closed 2021-04-13 by bdbcat