OpenCPN/OpenCPN #1394: SAR plugin missing in PPA distribution

dangowrt, 2019-05-05:
PPA distribution is missing the SAR plugin. It compiles and works fine from this source:
https://github.com/SaltyPaws/SAR_pi
I have build it locally for now, but other engineers will update the system which may lead to problems with that local build. Please include the SAR plugin in the PPA distribution!
We are running Debian 9 on the Sea Watch 3 SAR ship and have been heavily using this plugin in the past, no other software got that feature.

Thank you!

PS: If you need help having the software included in Debian official upstream, let me know, I've previously been helping zyGRIB to achieve that and it will actually make your lives easier (PPA can of course still be used for preview/early-access).

```
M/V SEA-WATCH 3
IMO №:          7302225
Call sign:      PE 7098
MMSI:           244 140 096
Phone V-SAT:    +49 30 1208 21911
Phone Iridium:  +88 167 770 7755
Web:            sea-watch.org
```

rgleason, 2019-05-05:
Are you aware of a user related issue regarding entry of the starting waypoint (degrees -decimal min vs degrees, min, seconds?)  that could cause problems in the tracks created? 
It is very important to use the correct waypoint format when copying and pasting from OpenCPN into SAR_pi.

Download of the plugin was canceled and a notice was posted for this reason.

We would like to see the plugin improved so that it can simply use an OpenCPN waypoint directly without having to copy and paste, and thus be assured it is the correct position!

Hakansv, 2019-05-06:
Rick..
Win10, O 5. sar_pi built from github master https://github.com/SaltyPaws/SAR_pi.git
I've tried this and have copied cursor position by Ctrl-S and used ship position with different O settings for the lat/lon format. In all cases was the position correct converted to decimal degrees in SAR_pi. See the shot where I've pointed on a WP and copied cursor position. I may misunderstand your message. What's the problem you refer to?
Håkan

[image]

rgleason, 2019-05-06:
Hakan, I should have included the Tracker reference.  This is what the problem is more definitively.

https://opencpn.org/flyspray/index.php?do=details&task_id=2110

Would be very interested in your response and thoughts.  I don't think the problem is fixxed yet but I might be wrong,  ... have been before)

Hakansv, 2019-05-06:
OK. I've tried three different sar patterns and using the lat/lon style failures reported in the flyspray post. I can't reproduce using Win10 O ver >5.0 (Present master). Also the tab ° ' " converts fine. Even W lon.
It may be O v.5 have changed the way it reports positions to the sar_pi? See the shot where all search pattern starts at desired WPs. ( The left down is the tab ° ' " conversion using "0" seconds.)
You better test as well?
Håkan

[image]

rgleason, 2019-05-06:
That's great news Hakan, I did not know the PR in Opencpn git was all that was needed. I thought additional work was needed at the plugin end!  

I will test it and I think Gille needs to test it too, since he found the problem (although I had noticed it earlier and unfortunately had not raised the flag). 

I will email him and post it on the thread.

rgleason, 2019-05-06:
Hakan, the bug is still there. Using Gille's first example, see this screen please. When I by hand enter the coordinates from the Waypoint that I made, into the Deg, Min, Sec and hit convert. What is shown is the result. It is not correct I am afraid.  I am using the Win ov50 version from the website page downloads.

[image]

Using the other Tab Decimal Degree and positioning the ship with "Drop ship here"
the lat / long are brought in reliably.

However
Using the Cursor feature which says highlight lat or long (in a waypoint properties) and hit ctrl + S  does not work. 

Waypoint entry should be seamless regardless of what units are being used in either Opencpn or SAR_pi,  and very reliable, it is a stressful moment and critical to guiding the search effort.  The software should just work without being finicky.

There are three ways to get the SAR start waypoint basically:
1.  Manual entry from Reports- VHF Call, Cell text, etc  (majority of SAR?)
2..  From current ship position, basically the MOB is from your ship but you have to guess how long ago and how far back in what direction, unless they have a transponder.  
3.  From an opencpn waypoint  - How does this happen?  You're trying to figure out where they fell off by locating a WP.

Using the Decimal Degree  "Ship" button works fine.
Using the Cursor does not work that well yet.
Why not have a cursor position then right click  and pick  "SAR Start Point" and it goes into the same box as the "Ship" button uses?

Also when the Search pattern is generated, it should be automatically labeled with the type of search  (Trackline, ExpSquare, Sector, Oilrig)  and Postion and added to the Routes.

leamas, 2019-05-08:
@dangowrt wrote:

> If you need help having the software included in Debian official upstream, let me know, I

If you have some secret way to make the FTP masters process this, let me know ;)
https://ftp-master.debian.org/new.html,  one of the first items....

rgleason, 2019-05-08:
Leamas this response may be attached to the wrong thread, but here is info on debian.
You can register as a debian member.  https://www.debian.org/doc/manuals/developers-reference/ch02.en.html#registering

Here is the master server https://www.debian.org/doc/manuals/developers-reference/ch04.en.html#servers-ftp-master

Here is how you upload to the servere
https://www.debian.org/doc/manuals/developers-reference/ch05.en.html#upload

leamas, 2019-05-08:
Hi Rick,

I do appreciate your enthusiasm. Really.

That said, you sometimes provide info which is not really useful. This is one of those occasions. I don't really see the point in starting a thread about the Debian packaging process.  Could we just please stop the  discussion on this particular subject here and now?

If you are really interested about the packaging process you need to investigate the roles of Debian Developers (a. k . a. DD), the FTP masters and the New queue. I don't really have the time to explain those things, sorry.

rgleason, 2019-05-08:
No problem. Understood. Was just trying to help re the dangowrt post.  I mis-understood...I'll just watch.

bdbcat, 2019-05-08:
@dangwrt...
We have been down the road of debian inclusion several time before.  Each time, we come to loggerheads with some reviewer insisting upon meaningless source code changes in order to pass some undefined test bar.  The main issue is usually this:  For inclusion, if we use any code that is available in a separate library for debian, we must use that library, however flawed it may be.  Our approach is usually to assimilate the library source code, fix the bugs, and move on to the next item. We thus decline their suggestions.  And there it stops. The debian gatekeepers have more interest in code purity than in user utility.  If you can find some sponsor who understands us, and our interest in app functionality for real users, please do.  We would try again.....

Dave

leamas, 2019-05-09:
We *are* trying as of now, and we have passed the review. The problem right now is the very last step, the 'new package' review performed by the FTP Masters. This is focused on licensing. 

For now it's just a long delay. I talked to the involved people recently. They said that historically three months isn't a long delay. I had my thoughts this, but kept them for myself..

dangowrt, 2019-05-09:
@Hakansv @rgleason Thank you for the healthy debate, let me know if there is anything I can help you with. Keep it going ;)

@bdbcat I understand that embedding libraries **is** a problem, not from the perspective of a single project, but from the perspective of  OS maintainers, yes, I do want to fight it. However, I also understand that it would be asking too much from a project like OpenCPN to have all their local patches included in upstream libraries **and** subsequently backport those changes to the version included in Debian. And this is exactly why I was asking this, because I can help here -- making sure patches fixing bugs in libraries get discussed and eventually merged into those projects, and making sure that Debian includes those patches as well. So I wanted to see that changes you have made for wx* and SQLiteCPP but there is no history of those libraries in your source tree. Hence I will need to diff against the vanilla version of the library, and for that'd it'd be useful to know the exact release you have been basing your local copy on.

@leamas Currently our usability of the OpenCPN project is limited by several factors: We are not using Ubuntu but (vanilla) Debian, and maintaining that PPA obviously also creates quite some work and effort is needed to keep it alive and updated, right now several previously included plugins got missing. Having the package included in Debian does greatly increase usability for users of Debian and Debian-based OS. I'm hence very glad to see your efforts in that process and will also poke around people which may hopefully accelerate it a bit more...

leamas, 2019-05-09:
@dangowrt: Any help to make the wheels turn around a little faster would of course be appreciated. Right now the problem seems to be that the Debian folks are in release freeze for Buster, making new packages a low priority.

However, in a longer perspective I'm have much more hope in this than Dave ;)

Also, in a longer perspective maintaining the plugins is a major obstacle. Packaging plugins is a big effort, and the current matrix of platforms X plugins is IMHO impossible to maintain. The logical solution would be to teach opencpn how to install it own plugins, like e. g.  browsers. This is the background to #1390 and #1115. I have no idea if I will be able to complete this, but it's a try.

leamas, 2020-10-18:
@dangowrt: @bdbcat: the opencpn-plugin-sar is now available in the ppa repo, the debian package is in place. Time to close this one?

dangowrt, 2020-10-18:
Yes, glad to see it works out-of-the-box now. Thank you all!

closed 2020-10-18 by dangowrt