OpenCPN/OpenCPN #1396: regression cmake sanitizer option is broken. and OSX warning

did-g, 2019-05-08:
Hi,
- cmake with sanitizer is broken.
`cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo DENABLE_SANITIZER=thread,undefined`
returns an error:

> CMake Error at CMakeLists.txt:1924 (target_link_libraries):
>   The keyword signature for target_link_libraries has already been used with
>   the target "opencpn".  All uses of target_link_libraries with a target must
>   be either all-keyword or all-plain. 

It was working before.

- there's a bunch of ld warnings in OSX build:

> ld: warning: direct access in function '_colTable::_colTable()' from file 'libSYMBOLS.a(chartsymbols.cpp.o)' to global weak symbol 'colorHashMap_wxImplementation_HashTable::DeleteNode(_wxHashTable_NodeBase*)' from file 'CMakeFiles/OpenCPN.dir/src/chart1.cpp.o' means the weak symbol cannot be overridden at runtime. This was likely caused by different translation units being compiled with different visibility settings

leamas, 2019-05-08:
Indeed... please try the attached patch for now, will make a PR later.
[0001-build-Fix-broken-link-command-1396.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3156904/0001-build-Fix-broken-link-command-1396.patch.gz)

leamas, 2019-05-08:
As for the warnings: although I'm not a maintainer, I think it's fair to say *Patches welcome*.

Actually, the visible patches are just the tip of the iceberg: we have disabled far too many warnings, notably unused variables. Filed as #1267.

did-g, 2019-05-08:
Thanks, it's working.

In the weird regression category:
weather_routing compiled outside O tree works fine in all cases.
inside O plugins subtree  after backporting:

diff --git a/src/ocpn_plugin.h b/src/ocpn_plugin.h
index c9e6c11..b604fd8 100644
--- a/src/ocpn_plugin.h
+++ b/src/ocpn_plugin.h
@@ -45,7 +45,7 @@
 #include <wx/xml/xml.h>

 #ifdef ocpnUSE_SVG
-#include "wxSVG/svg.h"
+#include <wx/bitmap.h>
 #endif // ocpnUSE_SVG

 #include <memory>
It compiles fine, but if O is configured without libELF it doesn't load.

Regards
Didier

leamas, 2019-05-10:
Didier,

It's great that you file bugs, appreciated! 

That said, if you could have one issue in each bug things would become easier, This bug is now three independent issues (the cmake failure, the  osx link problem  and the weather_routing stuff; it makes things needlessly complicated. 

It's just that I want to be able to close "my" bug once the patch is merged :)

rgleason, 2021-06-20:
Should this be closed now, as resolved?

closed 2023-02-02 by nohal