OpenCPN/OpenCPN #1398: way to pass speed/course waypoint information to plugins

doudou, 2019-05-10:
The plugin waypoint representation does not pass some of the OpenCPN internal waypoint information, such as speed and course. I was wondering whether there would be a way to get that information within a plugin already, or what it would take to extract it ?

Thanks

rgleason, 2019-05-10:
Isn't speed and course in the nmea data flow? It is not associated with a waypoint.

doudou, 2019-05-10:
The UI allows to set a per-waypoint desired course and speed. That's what I am looking for.

doudou, 2019-05-10:
For reference, it's available in the internal `Route` class, but not in `PlugIn_Waypoint`

doudou, 2019-05-10:
For now, I've hacked the plugin interface to add it straight to `PlugIn_Waypoint`. Given the versioned nature of OpenCPN plugin interface, I'm guessing that it is not the way ... Wondering if there's something I missed that would allow to get it, or for guidance on how to implement it properly to get it included in opencpn proper.

Forgot: OpenCPN version 5.0

nohal, 2019-05-10:
To keep the backward compatible API it must be done exactly the same way as `PlugIn_Position_Fix` and `PlugIn_Position_Fix_Ex`

juanman2, 2019-06-18:
@nohal  is this still a good first task?  If so, any hints about how to get started?

nohal, 2019-06-18:
Implementing `PlugIn_Waypoint_Ex` adding more data that may be available for a waypoint is technically a very simple first task, yes.
Why is it important for a plugin I can't assess as the actual use case is missing and the description is a little vague (for example there is nothing like a course for a waypoint, it, if anything, is the calculated direction to the next point in a route)

juanman2, 2019-06-18:
> On Jun 17, 2019, at 7:39 PM, Pavel Kalian <notifications@github.com> wrote:
> 
> a course for a waypoint, it, if anything, is the calculated direction to the next point in a route)

LOL … I was just composing an email to Doudou with exactly that question.  I saw a ‘planned SOG’, but only a computed COG, so ???

juan

juanman2, 2019-06-18:
@doudou It is my intention to work on this task as my first task.  Since it is an internal interface, I will need some way to properly test the solution, other than manually testing the UI.  Please send me a note with guidance as to how to proceed.

doudou, 2019-06-25:
> Why is it important for a plugin I can't assess as the actual use case is missing and the description is a little vague (for example there is nothing like a course for a waypoint, it, if anything, is the calculated direction to the next point in a route)

I am dealing with an autopilot system that can handle whole trajectories (read: splines) with velocity profiles. OpenCPN does not handle course (it's plain waypoints), but does allow to set velocities.

So, while within the current OpenCPN implementation the "course" is the direction to previous or next waypoint, this is not the case with the software I am interfacing opencpn with (which considers trajectories, not waypoints). Given the commitment of opencpn to not break existing interfaces, I thought it would be (much) better to do an extended Waypoint struct with both parameters in one go, instead of having speed now, and add course later.

doudou, 2019-06-25:
>  Since it is an internal interface, I will need some way to properly test the solution, other than manually testing the UI. Please send me a note with guidance as to how to proceed.

I don't know how OpenCPN is tested / whether it has automated tests. Maybe @nohal could give us some guidance there ?

rgleason, 2019-06-25:
No automated tests for Opencpn. You could look at the[ ocpn_draw api](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api:ocpn_draw_odapi),  in the Developer Manual  [Plugin Api section](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api) by Jon Gough, partially written up here (perhaps needs some updating.   It might give you some ideas...

juanman2, 2019-07-10:
Question .. is it possible to use Eclipse with OpenCPN? 

Juan

nohal, 2019-07-10:
What does "is it possible to use Eclipse with OpenCPN?" mean for you exactly? If you are asking if you can use Eclipse IDE to edit the code, there is nothing that can stop you. Cmake has 2 project generators that might make it easy - `Eclipse CDT4 - Ninja` and `Eclipse CDT4 - Unix Makefiles`. None of the regular contributors is using either as far as I can tell so how good is the experience is unknown.

scadaperspective, 2019-07-10:
I have used Eclipse quite a lot with OpenCPN and plugins over the course of it’s lifespan, from using it in debug mode to using eclipse mainly as an advanced text editor. 

I use Eclipse as a text editor ATM on  c++ with some extensions from the marketplace it really makes for a powerful editor. The new version has come a long way. I have not had time to try it out in debug mode yet.

As Pavel has said the experience varies, for java it is my preferred IDE.
I am yet to try it out with the latest version of QT

juanman2, 2019-08-11:
Sylvain, Pavel:

I apologize for going dark mode. I had much to do on the boat and a lot to take care off to modernize my Emacs setup. I’m ready to code. 

Juan

juanman2, 2019-08-13:
Sylvain:

A few questions:

Looking at your code you seem to be using a callback std::unique_ptr<PlugIn_Waypoint> GetWaypoint_Plugin( const wxString& GUID) from the plugin to the main program to obtain the RoutePoint by GUID.  I presume that you have the GUID because you got the Route object earlier via one of the messages sent from the main program.   

(1) This makes sense to me, since it is clear you need a RoutePoint data object, not a WayPoint or a Position.  
(2) I presume your autopilot wants to be initialized with a message that says something like [(<bearing>, <speed>), (180, 6), (210, 7), (250, 3), (nil, nil)] .. or some such? ie. All the pairs of bearing/speed for the complete route in one message, correct? 
(3) How are you obtaining the RoutePoint GUID?  What message are you getting from?

Thanks,

Juan

doudou, 2019-08-19:
Hi @juanman2. Sorry for taking so long to reply. Lots on my plate, this kind of went to the back burner.

> (2) I presume your autopilot wants to be initialized with a message that says something like [(<bearing>, <speed>), (180, 6), (210, 7), (250, 3), (nil, nil)] .. or some such? ie. All the pairs of bearing/speed for the complete route in one message, correct?

It actually wants position, course and speed. It does not work using a target bearing, but as a cartesian trajectory. See https://github.com/tidewise/gui-seabots_pi/blob/master/src/OCPNInterfaceImpl.cpp

rgleason, 2019-08-19:
You asked above about automated tests.  There are none as far as I know, but some programmers here are using these tools

 Testquality  - to record issues, set up and define necessary tests, and to record those tests. estQuality is completely FREE for use with GitHub public repo’s  http://www.testquality.com/github-test-management/

Codacy - Standardize code, clean up code and find errors.  Free to opensource  https://www.codacy.com/
Love those test reports  https://github.com/canne/dashboard_tactics_pi/releases/tag/v0.4.1

and of course appveyor and travis

juanman2, 2019-08-30:
@doudou  Looking ... I'm trying to understand what structures need to be expanded and versioned and how those are sent from the main program to the plugin.

bdbcat, 2021-04-29:
Closing, inactive

closed 2021-04-29 by bdbcat