OpenCPN/OpenCPN #1399: Error during build - ArchLinux

mcbloch, 2019-05-12:
Build is broken when clean building OpenCPN on arch from [AUR: opencpn](https://aur.archlinux.org/packages/opencpn/). 

I have the same issue as ve7it (last edited 2019-04-29 01:55) explains in the comments of the package.
My build log gives the same output as his which can be found here: http://ve7it.cowlug.org/build.log

nohal, 2019-05-12:
Nothing to do with OpenCPN, you are missing parts of wxWidgets, at least AUI and OpenGL support.

closed 2019-05-12 by nohal

mcbloch, 2019-05-12:
@nohal i'm looking at how I should fix this but can't find the missing parts. Do you have an idea of where I should look? Could use some help here :)

nohal, 2019-05-12:
I'm not an Arch user and have no idea how the wx packages are built there, so sorry, I don't. You can sure build it from source instead. I kind of thought while creating an AUR the author is supposed to know what he is doing and the purpose of all that is to make it reproducible for others.
Also be very careful about GTK-3, Wayland and similar "modern" stuff with wxWidgets 3.0, that it builds does not at all mean it works.

taruti, 2019-05-13:
Does using wxgtk2 work for you? `pacman -S wxgtk2`? And then in the PKBBUILD change
```
  cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr \
        -DBUNDLE_GSHHS=CRUDE -DBUNDLE_TCDATA=ON -DBUNDLE_DOCS=ON \
        -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-gtk3 \
        -DOCPN_FORCE_GTK3=ON
```
to
```
  cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr \
        -DBUNDLE_GSHHS=CRUDE -DBUNDLE_TCDATA=ON -DBUNDLE_DOCS=ON 
```

I'm running and compiling master successfully on Arch Linux, but linked to wxgtk2 instead of wxgtk3.

mcbloch, 2019-05-14:
Clean building wxWidgets like nohal said did not fix the issue.

I then installed wxgtk2 and updated the PKGBUILD like you said @taruti, but this does not seem to change the dependency of the build. It still crashes trying to compile with the gtk3 version.

nohal, 2019-05-14:
With the mess your system must be now, it is hard to say what is used and what "crashes", so let's start to try to find out.
You built wxWidgets from source, good, how exactly? Where exactly it is installed on your machine now? Does the build "crash" because of your local build of wxgtk being done against gtk3 and incomplete or because it keeps using the packaged wxgtk3? Do you need wxgtk3 package for anything else?

mcbloch, 2019-05-15:
I donwloaded the sources from their website and build it using [these](https://github.com/wxWidgets/wxWidgets/tree/master/docs/gtk) instructions.
I have 2 install atm (seen by executing wx-config --list), located at:
- /usr/lib/wx/config/gtk2-unicode-3.1
- /usr/lib/wx/config/gtk3-unicode-3.0
I'm unsure wheter the gt3 one is the one I build myself or not.

I tried to install opencpn again to see on what wxgtk build it crashed and it completed. I see that the patch from issue 1329 got applied in the aur, I probably did it wrong when I did it myself then.

Thanks for your help!