OpenCPN/OpenCPN #1401: Creating package on Travis

nohal, 2019-05-15:
After migration of the travis build to Mojave machines with Xcode 10.2 (required for the dark mode support in the automatic builds) the `create-dmg` script partly fails with AppleScript timeout resulting in the dmg not having the correct geometry and background:

```
Running Applescript: /usr/bin/osascript "/var/folders/17/5mc7816d3mndxjqgplq6057w0000gn/T/createdmg.tmp.XXXXXXXXXX.PLSoOHC5" "OpenCPN Installer"
/var/folders/17/5mc7816d3mndxjqgplq6057w0000gn/T/createdmg.tmp.XXXXXXXXXX.PLSoOHC5:405:417: execution error: Finder got an error: AppleEvent timed out. (-1712)
Done running the applescript...
```
If it won't disappear by itself, we will probably have to do production builds locally or try to use some alternative solution for the DMG creation

- https://github.com/sindresorhus/create-dmg
- https://github.com/LinusU/node-appdmg
- https://pypi.org/project/dmgbuild/

bdbcat, 2021-04-30:
1.  Official distribution is .pkg installer
2. We are migrating away from Travis.

So, OK to close?

nohal, 2021-04-30:
Yeah, I think it's time...

closed 2021-04-30 by nohal