OpenCPN/OpenCPN #1402: OSM BE/NL support

mcbloch, 2019-05-15:
I want to use the OpenSeaMaps charts in OpenCPN but the belgian or dutch OSM charts provided as download via the ChartDownloader plugin seem to be very insufficient. 

I don't see any object of interest such as buoys, traffic routes, etc.

Selecting the most options possible in the Chart settings doesn't seem to show anything more.

Am I doing something wrong with the download or are the provided raster charts just like this. Does this mean I need to somehow make my own charts or can you download more detailed ones online?

nohal, 2019-05-15:
This is what OpenSeaMaps is, those are no navigation charts. For the inland waterways of both BE and NL, you can get the free iENC charts through the chart downloader, for the adjacent sea, your best bet is https://o-charts.org/shop/index.php?id_product=10&controller=product&id_lang=4

As this is not an OpenCPN issue, I'm closing it.

closed 2019-05-15 by nohal

mcbloch, 2019-05-15:
If you look [here](http://map.openseamap.org/) you get to see the full openseamaps charts. These are very detailed and have good navigational information. Because of this I think that the downloader is not downloading the most detailed and relevant OSM maps.

nohal, 2019-05-15:
The downloader is downloading exactly what the catalog points to. Neither the catalogs nor the archives it points to are created or maintained by OpenCPN project. This issue is completely irrelevant here.

nohal, 2019-05-15:
The correct place to complain in this particular case is https://www.openwaterkaart.nl as you can see from the catalog.

mcbloch, 2019-05-15:
Thanks for the assistence for my issue! I'll see what I can do there.