OpenCPN/OpenCPN #1403: AIS feed by network chookes the system

Hakansv, 2019-05-17:
Win 10 present master "Sandbox". An observation for info.
To receive a massive flow of AIS targest via Internet was before possible without choking the system to much.
Testing the same on present code i get the feeling things have gone worse. Now O is nearly not answering using the same connection as before. This note is for info if someone changed the connections code recently.
These are my connection parameters if you want to test the same: (For any reason I can't understand the only protocol usable for this site is GPSD on Windows! That's nothing new. Have been like this for years)
[image]

transmitterdan, 2019-05-17:
@Hakansv 

My experience with this AIS feed is that the number of vessels varies from a few thousand to a ridiculous number.  I have seen >25K vessel reports on occasion.  Anyway, it is unlikely the connection code is the problem.  It is more likely to be the AIS list handling and rendering of targets that must be updated as each vessel report arrives.

Hakansv, 2019-05-17:
@transmitterdan 
Yes, I know. I've been using this site as a test for years. This note was for the purpose someone has changed anything related since now the rendering was disturbed in a way I haven't seen before. But it can as a well be the site data flow is more extensive than ever. 
Thanks/ Håkan

rgleason, 2019-05-17:
Can't you limit the range in Opencpn?

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais&s[]=ais#ais_target_list

Limit Range: The list will only show targets with a range less than, or equal to this value.

Hakansv, 2019-12-30:
No solution and it's not a "real" problem. Only relevant for a test environment, not while cruising.

closed 2019-12-30 by Hakansv