OpenCPN/OpenCPN #1406: Semantic of plugin LateInit

did-g, 2019-05-18:
Hi,
Plugin LateInit is not called after a plugin is switched from disable to enable in options box but only on startup. Should it be? Or should it be call for all plugin after options box is closed?

For example DRAW is using it for saying it's ready for business.
BTW are other plugins setting WANTS_LATE_INIT ?
 
> commit 93b2a07f3adc40318a347863fcd2917790af2cfd
> Date:   Thu Jun 6 16:09:28 2013 -0700
> 
>     Add LateInit() as optional plugin functionality.
>     
>     LateInit() is called if a plugin returns WANTS_LATE_INIT of its Init()
>     call.
>     
>     LateInit() is called when all other initialization and setup has been
>     done, including first graphical update of the application. This enables a
>     plugin to place itself on top and gain focus if needed.
>  
>

rgleason, 2021-04-23:
@jongough @bdbcat  Comments?  Didier has some point. What is appropriate?

bdbcat, 2021-04-29:
Called only on startup, as designed.
AFAIK, only ODraw uses this function....

closed 2021-04-29 by bdbcat