OpenCPN/OpenCPN #1408: Dashboard clock displays cpu time instead of GPS time when there is an active route.

rgleason, 2019-05-24:
LifePart2 reports and Moo has confirmed

[FS#2568 - Dashboard clock displays cpu time instead of GPS time when there is an active route. ](https://opencpn.org/flyspray/index.php?do=details&task_id=2568)

Refer to cruiser forum thread
[Re: no VMG, HDG, COG on Dashboard (plus other issues)  Post #38](http://www.cruisersforum.com/forums/f134/no-vmg-hdg-cog-on-dashboard-plus-other-issues-217998.html#post2894834)

BerndCirotzki, 2019-05-24:
Hello rgleason,
i think, that the problem ist also, that when a route is aktive, it seams, that there is no NMEA Data receiving from the COM Port, where OpenCPN sends the route data to. So not any Plugin is now getting NMEA Data from this Port! (the Dashboard now knows nothing.)
I tryed my autopilot to go to a route waypoint that was directly in front of my ship, but when a started the track mode by using the route my ship made a big turn ! So a this there is a BIG BUG in the "aktiv route" feature ! in 4.x Version it works.
I wrote this Issue on tracker OPENCPN.org Webpage.

transmitterdan, 2019-05-24:
I confirm there is a problem.  I will investigate.

rgleason, 2019-05-24:
BerndCirotzk Tracker
[FS#2552 - No receiving NMEA data from com port when a route is aktiv ](https://opencpn.org/flyspray/index.php?do=details&task_id=2552&string=+BerndCirotzki+&search_name=&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=&order=openedby&sort=asc&pagenum=2)

transmitterdan, 2019-05-28:
So the problem with the clock is this:

When a route starts running then OpenCPN begins emitting ECRMC sentences.  These messages contain the local CPU UTC time. Eventually these messages get passed up to all plugins and Dashboard thinks they are valid RMC messages.  So it displays the local CPU UTC time until it gets a different RMC message from a GPS for example.  Then the clock jumps to that UTC time and so on.  So the clock will jump between local CPU UTC time and GPS UTC time depending on how often it receives GPS or ECDIS time.

It is fairly trivial to make it so Dashboard will ignore the time stamp from the ECRMC sentence.  Is that an acceptable solution?  If so, should such ignore be done via a user flag in the preferences or just ignore time stamps from ECRMC altogether?

I cannot find the relation to the other claim of stopping  reception on a serial port during a route follow.  I cannot duplicate that problem so far.  I think the serial port issue and the clock issue are unrelated.

rgleason, 2019-05-28:
Yikes, how did you figure that circuitous RMC route out! I'll let others comment on the option or  not.

rgleason, 2019-08-25:
Did this get fixed? I thought TDan fixed it. Should it be closed?

rgleason, 2019-10-12:
I asked about this on [CF  Plugin:Dashboard thread](http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087.html#post2993382) 

Please see [Post #537](http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087.html#post2994794) 

> Seeing Dan's explanation about the root cause being the OpenCPN generated RMC sentence I would get the O fix first done for it and the close the Dashboard ones which are duplicates anyway. If the fix is to not distribute OpenCPN generated RMCs to plug-ins anymore, they would lose the VMG-to-the-next-WP information in that sentence (remember, the non-Tactics "VMG"?) as a secondary effect. Since I am not authoring either of the modules and do not plan to do so, this is only my personal opinion, since you asked for it

transmitterdan, 2019-10-14:
PR #1468 should solve this issue.

rgleason, 2019-11-14:
OK, Closing

closed 2019-11-14 by rgleason