OpenCPN/OpenCPN #1409: Mac: Installation paths includes build environment

leamas, 2019-05-25:
In Mac, the result of *make install* seemingly includes the build environment. For example, one path looks like

    oesenc_pi-1.2.0-1_darwin-10.13.3/Users/travis/build/leamas/oesenc_pi/build/OpenCPN.app/Contents/PlugIns/liboesenc_pi.dylib rep

Obviously, this makes no sense. Having elements from both the travis build user and my userid creates serious problems to have reproducible results. This  might have slipped through the net because the only artifact from the mac build so far has been the dmg file, and this does not siffer from this.

A simple *make install* on a macos machine should reveal the complete truth....

nohal, 2019-05-25:
`make install` of what? If of OpenCPN, then no, this is not what happens.

leamas, 2019-05-25:
sory, my bad. This is about oesenc_pi plugin, and should be filed against that.

I'm a bit worried that each and every plugin have different macos install paths. The rules  I use now on oesenc_pi is a bunch of heuristics. The install paths should be fixed, but I don't know this macos installation stuff well enough to be able to do that.

Sigh.

Closing

closed 2019-05-25 by leamas

leamas, 2019-05-25:
Filed https://github.com/bdbcat/oesenc_pi/issues/37