OpenCPN/OpenCPN #1412: Preferences window has incorrect background colour on macOS

caesar, 2019-05-31:
OpenCPN 5.0.0
macOS 10.14.5

In light mode on macOS, the background colour of Preferences window is `#aaaaaa`.
It should be `#ececec`.

Interestingly, it is fine in dark mode, and fine in light mode after switching into dark mode and out again. Weird.

[image]

The status bar colours  are a bit strange too… but that's for a separate issue I guess.

nohal, 2019-05-31:
The whole diming logic is disabled in dark mode for now as we are simply not able to do a better job than the OS, at least not without major effort...

caesar, 2019-05-31:
Hmm, interesting. No, as you say, not much need for it in dark mode (for UI elements that is).

But here I have dimming turned off and am on the default colour scheme – so far as I can tell… I've cycled through all of them, anyway.

The issue is the same with all dialogs BTW, not just Preferences.

nohal, 2019-05-31:
Yes, `DimeControl` is called on everything. And even if the minor issues it has are fixed, it still won't be able to be 100% good due to the limitations of wxWidgets - we simply can't set colors for everything.

caesar, 2019-05-31:
Is there a way it could be disabled if dimming is not turned on (or the colourscheme is the default),  just as it is for dark mode?

Failing that, at least hardcoding a few important colours like the dialog background colour and primary text colour would go a long way to making it look better.

I can't remember how it was implemented before – I remember fixing some colour issues – but the colours looked reasonably okay in previous versions IIRC.

nohal, 2019-05-31:
Nope, but of course it would need just a little change in https://github.com/OpenCPN/OpenCPN/blob/4f76ba5d8c59a348a47283f67f079630c1aa5605/src/navutil.cpp#L4672

The biggest issue that complicates the so appealing simple solution of doing nothing in the default colourscheme is that the already created windows would stay "wrong" if you cycled back to the default from another one.

caesar, 2019-05-31:
> the already created windows would stay "wrong" if you cycled back to the default from another one.

Aaaahhhh… yes, of course. Hmmmm… The same problem exists when entering Dark Mode after OCPN is open – it looks weird. Pity, because it looks great if Dark Mode is enabled before it's launched.

Honestly for me I think a better solution than the current system would be only to have Dark Mode and Light Mode, and forget about dimming… but that may not be popular with other people :-)

caesar, 2019-05-31:
Anyhow, fixing a few default colours is probably a simple start, even if it's not an ideal solution.

caesar, 2019-05-31:
The About dialog has (partly) correct colours. But the links have a hardcoded colour and don't show up in Dark Mode.

nohal, 2019-05-31:
Well, probably the closest to an ideal is to remember the original color assigned to each element type by the OS/wxWidgets and use that in the default colourscheme. Not surprisingly also more laborious :) With the links in the About dialog I vaguely remember there is nothing we can do, at least without reimplementing the wxWidgets controls...

nohal, 2019-05-31:
BTW, there is also a bug, mine, that needs to be fixed related to this - picking up the font colour from the OS was a brilliant idea for the UI, but it really sucks having white text for the vector chart in dark mode...

caesar, 2019-05-31:
> it really sucks having white text for the vector chart in dark mode...

Haha, yeah, that wouldn't be great!
I'm not seeing it on CM93 though, is it only S63? (Haven't installed the plugin on O5 yet.)

caesar, 2019-05-31:
In general the colour support is much better in Dark Mode… fully explained by the fact that dimming is disabled and native colours are used :-D

nohal, 2019-05-31:
> I'm not seeing it on CM93 though, is it only S63? (Haven't installed the plugin on O5 yet.)

You will if you start with a clean table without `opencpn.ini`, I was insane while doing that, but not that insane...

caesar, 2019-05-31:
Aha, I'll try that… probably a good idea before filing any bugs anyway 😆

caesar, 2019-06-02:
It's worse than I thought. It's not just an incorrect background colour; all the backgrounds are weird. I think the whole of `DimeControl` needs reworking.

The background of the Object Query dialog actually gets *lighter* in the Dusk and Night colourschemes!

In this image, the first row is Daylight, the second row is Dusk, and the third row is Night:

[image]

caesar, 2019-06-02:
Proposal:

- In **Default** and **Daylight** modes, put the whole app into macOS's **Light Mode** and use the _unmodified system colours_.

- In **Dusk** and **Night** modes, put the whole app into macOS's **Dark Mode** and use the _unmodified system colours_.

- If the system-wide preference is Dark Mode, use Dark Mode UI in Default and Daylight too.
However, never use Light Mode UI in Dusk and Night modes, regardless of the system-wide setting.

Thoughts?

nohal, 2019-06-02:
- There is no dark mode on pre-Mojave. And some people do use pre-Mojave macOS versions with OpenCPN.
- There is not just macOS (yes, the current implementation works worst in macOS daylight mode, but part of it is wxWidgets and the OS itself)

caesar, 2019-06-02:
True Pavel (I was just editing my comment with a note to that effect).
Of course our own adjustments will still be necessary on other platforms.
It would still be nice to improve it (for other platforms too).

nohal, 2019-06-02:
And I don't know if we can switch between light and dark mode at application level at runtime, but obviously the time I have invested in researching that is exactly nil.

caesar, 2019-06-02:
> And I don't know if we can switch between light and dark mode at application level at runtime

I don't know if we can from wxWidgets… normal macOS apps can do so by changing [`NSApplication.appearance`](https://developer.apple.com/documentation/appkit/nsapplication/2967170-appearance?language=objc).

caesar, 2019-06-02:
Looks like you've already implemented that here as "use dark window decorations":

https://github.com/OpenCPN/OpenCPN/blob/83a3c4b5ff011d4eb070f009e0a46d194b177047/src/DarkMode.mm#L51

nohal, 2019-06-02:
That never affected anything but the window titlebar. At the time of implementation that is.

caesar, 2019-06-02:
No… but (I'm _fairly_ sure) it's the same setting that on Mojave was broadened to affect all controls. Only in OCPN we are overriding it for every control…

caesar, 2019-06-02:
Okay, I've got app-level dark mode working with
```objective-c
[[NSApplication sharedApplication] setAppearance:[NSAppearance appearanceNamed:NSAppearanceNameVibrantDark]];
```
I'll play with this a bit more over the next day or two as time permits and see if I can come up with something workable.

By the way does OCPN support Dark Mode on Windows 10 in any way yet?
I'm guessing that's harder for a wxWidgets app, but haven't looked into it at all.

caesar, 2019-06-02:
[image]

nohal, 2019-06-02:
> By the way does OCPN support Dark Mode on Windows 10 in any way yet?
> I'm guessing that's harder for a wxWidgets app, but haven't looked into it at all.

No idea, all the W10 machines I can get access to are running too old releases without dark mode support.

caesar, 2019-06-02:
I'll try it on my newly-updated Win10 v1903 VM later/tomorrow :-)
(My main navigation computer is actually Win10 so that's more important to me than Mac to be honest.)

I have the colours looking pretty good on my mac now, a bit more testing and tweaking required and then some testing on Windows and Linux…

Do you have any old versions of macOS / OS X to test on…?

rgleason, 2019-06-02:
https://helpdeskgeek.com/windows-10/how-to-enable-dark-mode-everywhere-on-windows-10/

caesar, 2019-06-02:
Okay so there's zero Dark Mode support on Win10. Hardly surprising, it's very new.

Current colours are pretty bad in the dusk and night colourschemes on Windows, too… hopefully I can improve that as well as improving the Mac colours.

[image]

nohal, 2019-06-03:
Looks even worse than it was with wx2.8 when we last actually tried to make it at least somewhat acceptable. I'm kind of worried that while it sure can be made "better", it can't be made "good" without extensive patches to wx..

caesar, 2019-06-03:
@nohal, can you remember what this routine was for…?

https://github.com/OpenCPN/OpenCPN/blob/83a3c4b5ff011d4eb070f009e0a46d194b177047/src/DarkMode.mm#L27

It doesn't seem to make any visible difference when I disable it, on macOS 10.13 (and 10.14, but maybe it wouldn't).

PR #1420 (2019-11-22): Improve UI colours and macOS dark mode support
Improves UI colours for dialogs etc cross-platform, especially on macOS.

- The most-improved platform is macOS 10.14+, where native light and dark modes are now used unmodified.
- Older macOS versions are also much improved.
- Windows 10 is improved, but still far from perfect.

Fixes #1412 – although I would like to do more work on this at some point, potentially fully reworking `DimeControl` and the colour-changing logic in general.
closed 2019-11-22 by bdbcat