OpenCPN/OpenCPN #1413: Enter Full Screen menu item appears twice on macOS

caesar, 2019-05-31:
OpenCPN 5.0.0
macOS 10.14.5

The "Enter Full Screen" menu item in the View menu appears twice.

The first one activates 'old-style' full screen, where OCPN take control and draws itself on top of everything. There is no way back out except via the context menu.

The second activates macOS native full screen.

IIRC, when I implemented the menus we were using a version of wxWidgets which didn't  support native full screen.

I suspect that the first entry is the one we are adding, and that wx is adding the second one itself. This could be resolved by simply removing our own entry.

Further to that though, if activated from, say, the context menu, we should also defer to the native macOS implementation instead of taking over the screen ourselves.

I'll see what I can do and hopefully submit a PR in the next few days.

bdbcat, 2021-05-01:
I merged your PR, resolving simple conflicts easily.  Tested on Catalina, seems OK with simple smoke test.  Tested on Mojave, which seems a little different.  Here the full screen must be exited by context menu.  But it basically works. Not tested any plugins.
So, time for more extensive testing....bring it on...

caesar, 2021-12-18:
This seems to be working fine in the current release; sorry for not following up earlier. I'll close this issue.

PR #1417 (2021-05-01): Fix fullscreen on macOS
- Remove the duplicate Enter Full Screen menu item on macOS.

- Enable native full screen method when full screen is triggered from context menu etc.

Fixes #1413
closed 2021-12-18 by caesar