OpenCPN/OpenCPN #1414: All menu shortcuts have disappeared on macOS

caesar, 2019-05-31:
This is just weird. Must be a wxWidgets issue I think.

The only items which have shortcuts are those added by the system / by wxWidgets.

[image]

caesar, 2019-05-31:
Sorry for the spate of issues all at once! There are more to come I think.
I'll do what I can to fix them though ;-)

caesar, 2019-05-31:
This is caused by this conditional check disabling keyboard shortcuts if `g_bresponsive` is true.

https://github.com/OpenCPN/OpenCPN/blob/83a3c4b5ff011d4eb070f009e0a46d194b177047/src/chart1.cpp#L5542

If I comment out the specified line, keyboard shortcuts work on macOS.

I'm not sure what that check is for; I'll do some further research.

caesar, 2019-05-31:
Okay, so `g_bresponsive` seems to mean that "Scaled Graphics interface" is enabled.

I have this enabled in 5.0 because I need to make the toolbar icons bigger; otherwise they're really small on my Retina screen.

The question is why it would be a reason to remove the keyboard shortcuts from the menus… 
@bdbcat @nohal any ideas?

caesar, 2019-05-31:
So far as I can see the `if(!g_bresponsive)` condition mentioned above is being used as a proxy for "if not Android"… if so, if it would be better to replace it with an `#ifndef __OCPN__ANDROID__`.

caesar, 2019-05-31:
Looks like that line was introduced by Dave in 5452aab.
It does *seem* to just be a way of checking if we're on Android, and it looks like it was only intended to affect the context menu.
I'm going to go ahead and replace it with `#ifndef __OCPN__ANDROID__` and submit a PR.

bdbcat, 2019-11-22:
Ceasar...
g_bresponsive is not a proxy for "not Android".  I honestly cannot recall why this condition is used.  Maybe due to some layout considerations on some platforms?

Android has no Menu that can be managed by wxWidgets, in the desktop GUI sense.

caesar, 2019-11-22:
Hi Dave, I know that's not what it *means*, but it seemed to be being used that way here... At any rate it was added in a commit where everything else was for Android.
Anyway, it's doesn't seem appropriate to disable all menu shortcuts just because scaled graphics are enabled, as it's necessary to enable scaled graphics on most modern machines with high dpi screens.

I'm going to be offline for the next three months, but I'll come back to this and other issues when I'm back.

By the way thanks for merging a couple of my other issues from earlier this year.

PR #1418 (2019-12-07): Fix missing menu shortcuts with Scaled Graphics enabled
Instead of disabling menu shortcuts if `g_bresponsive` is true, only disable them on Android.

I believe this fix is appropriate. See #1414 for further discussion.

Fixes #1414
closed 2019-12-07 by bdbcat