OpenCPN/OpenCPN #1416: Add .editorconfig

caesar, 2019-05-31:
An [editorconfig](https://editorconfig.org/) file would help maintain coding style (at least for those who use supporting editors, or editors for which supporting plugins are available).

nohal, 2019-05-31:
We also have #376 and #1118 and the inevitable fun connected with running `clang-format` on everything. But there will never be a "good" moment for that.

caesar, 2019-05-31:
Yes… it's really hard to retroactively enforce a code style on old code!
It would be great to be able to have some linting set up though…

leamas, 2019-06-03:
>  ... and the inevitable fun connected with running clang-format on everything But there will never be a "good" moment for that.

Indeed.  However, in hindsight the time after the 5.0.0 release was perhaps a less awful time than now.

BTW: The main problem with such a major change is the open PR:s, right? One question is then why even simple bugfix PR tends to take quite some time before merged. Is this a strategy, or just a  maintainer ENOTIME error condition?

nohal, 2019-06-03:
the problem are the changes that do not even have a PR. And yes, I personally currently have zero time available.

leamas, 2019-06-03:
> the problem are the changes that do not even have a PR. 

Well,, it's hard for me  to fathom what tthat would be...

> nd yes, I personally currently have zero time available.

Fair enough.   Which also means that the unmerged  "simple" PRs ere unmerged  due to  ENOTIME rather than a strategy, right?

bdbcat, 2019-06-04:
Correct.
Dave

leamas, 2021-09-02:
Anyway, we have now re-formatted all sources in #2357. Adding a .editorconfig seems like a logical next step .

PR #2406 (2022-04-28): Add editorconfig
A bit lazy here, bundling two unrelated patches in one PR. Please cherry-pick if you want, the editorconfig file could wait to after 5.6.0.

EDIT:
Closes: #1416 
Closes: #2405
commit 93e999f9 (2022-04-28): Add an initial editorconfig file (#1416).

Closes: #1416
closed 2022-04-28 by nohal