OpenCPN/OpenCPN #1419: Object Query dialog always opens nearly maximized

caesar, 2019-06-02:
OpenCPN 5.0.0
macOS 10.14.5

### Expected behaviour

The Object Query dialog should default to a sensible size.
If resized, maybe it should remember the previous size when opened again. (?)

### Actual behaviour

The Object Query dialog always opens covering almost the whole screen, regardless of what size it was previously resized to.

### Screenshot

[image]

caesar, 2019-06-03:
Meanwhile, on Windows 10, it opens much too small by default, but it does remember the previous size.
[image]

rgleason, 2021-04-25:
Windows 10  Release Version OpenCPN 5.2.4+6b314e6
No longer "Remembers" the previous size of an Object query.
So it is a problem with Windows  OpenCPN 5.2.4 as well.

[image]

It would be useful to have the value remembered from the previous use!  Like this.

[image]

bdbcat, 2021-05-10:
Closing, corrected by commit.
Please re-open if necessary.

closed 2021-05-10 by bdbcat