OpenCPN/OpenCPN #1422: Dashboard_pi  VMG does not work, missing code

rgleason, 2019-06-08:


rgleason, 2019-06-08:
Ran a VDR_pi file with this which has VMG=6.493 towards the end.
Dashboard does not show this value.

$OCRMB,A,0.000,R,001,002,4545.029,N,01333.541,E,3. 997,4.308,6.493,V,A*67

For this VMG instrument, it is in Dashboard's responsibility to extract the VMG out of it (easy, done by the NMEA class) and send it to ID_DBP_I_VMG and ID_DBP_D_VMG type of instruments it has created. The sentence type shall be OCPN_DBP_STC_VMG.

Dead simple. Somebody must have deleted that code from the Dashboard at one point.

transmitterdan, 2019-06-08:
The checksum is not valid so O discards the message.  Thus Dashboard never sees the message.

transmitterdan, 2019-06-08:
Even with a valid checksum it doesn't drive VMG instrument.  The issue is that RMB sentence is not processed at all by Dashboard.  I can fix that.
What about VTG sentence.  Should it drive VMG instrument? If so, what priority should VTG have over RMB?

canne, 2019-06-08:
Hi, the checksum actually fails because of a cut and paste since "3. 997" should be "3.997", in which case the *67 is OK. But the sentence fails indeed in rmb.cpp because of a checksum failure. Why? Because the class is pre NMEA v2.3 and the sentence (generated by ov50 routing, BTW) is according to OpenCPN documentation https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:nmea_sentences#rmb_-_recommended_minimum_navigation_information - I fixed the class like this: https://github.com/canne/dashboard_tactics_pi/commit/46b953e593c0184236b1af045bbd46f1c88e4150#r33862376 .

I have successfully tested the two VMG instruments  three different boat's data when writing something like this:

```
        else if (m_NMEA0183.LastSentenceIDReceived == _T("RMB")) {
            if (m_NMEA0183.Parse()) {
                if (m_NMEA0183.Rmb.IsDataValid == NTrue) {
                    if ( !std::isnan(m_NMEA0183.Rmb.BearingToDestinationDegreesTrue) &&
                         (m_NMEA0183.Rmb.BearingToDestinationDegreesTrue < 999.) ) // empty field
                        SendSentenceToAllInstruments(
                            OCPN_DBP_STC_BRG, m_NMEA0183.Rmb.BearingToDestinationDegreesTrue, m_NMEA0183.Rmb.To);
                    if ( !std::isnan(m_NMEA0183.Rmb.RangeToDestinationNauticalMiles) &&
                         (m_NMEA0183.Rmb.RangeToDestinationNauticalMiles < 999.) ) // empty field
                        SendSentenceToAllInstruments(
                            OCPN_DBP_STC_DTW, m_NMEA0183.Rmb.RangeToDestinationNauticalMiles, _T("Nm"));
                    if ( !std::isnan(m_NMEA0183.Rmb.DestinationClosingVelocityKnots) &&
                         (m_NMEA0183.Rmb.DestinationClosingVelocityKnots < 999.) ) { // empty field
                        SendSentenceToAllInstruments(
                            OCPN_DBP_STC_VMG, toUsrSpeed_Plugin(
                                m_NMEA0183.Rmb.DestinationClosingVelocityKnots, g_iDashWindSpeedUnit ),
                            getUsrSpeedUnit_Plugin( g_iDashWindSpeedUnit ) );
                        this->SetNMEASentence_Arm_VMG_Watchdog();
                    } // then valid sentence with VMG information received
                    if (!std::isnan(m_NMEA0183.Rmb.BearingToDestinationDegreesTrue) &&
                        (m_NMEA0183.Rmb.BearingToDestinationDegreesTrue < 999. ) ) {
                        SendSentenceToAllInstruments(
                            OCPN_DBP_STC_BRG, m_NMEA0183.Rmb.BearingToDestinationDegreesTrue, m_NMEA0183.ErrorMessage);
                        this->SetNMEASentence_Arm_BRG_Watchdog();
                    } // then valid bearing destination
                } // then valid data
            } // then sentence parse OK
        } // then last sentence is RMB
```
You can see that I have added a watchdog for VMG - it is necessary because ov50 routing function is sending the above sentence when there is an active waypoint. This way, when one deletes the acive waypoint, sentences stop and after a timeout the two instruments are going to zero - for that I had to add an extra check in dial.cpp SetData() since it gets stuck to last data it receives.

For this purpose of showing VMG to the acive waypoint, the above mods are, IMO, enough. No reason to add VTG as _STC_VMG source. I can see how it can drive that sentence:
```
$GPVTG,t,T,,,s.ss,N,s.ss,K*hh
1    = Track made good
2    = Fixed text 'T' indicates that track made good is relative to true north
3    = not used
4    = not used
5    = Speed over ground in knots
6    = Fixed text 'N' indicates that speed over ground in in knots
7    = Speed over ground in kilometers/hour
8    = Fixed text 'K' indicates that speed over ground is in kilometers/hour
9    = Checksum
```
"Track made good" - oh, that's confusing. If both RMB and VTG are coming in (I am sure that it is possible) and used for the same purpose, the program should ask from the user what she prefers. I cannot see how this decision can be taken by a hard coded priority. I would leave VTG out until somebody asks for it for this particular purpose..

rgleason, 2019-06-09:
Thank you Petri and Dan.
We have had many discussions about the meaning of VMG (This one is VMG to waypoint,  and uses the common term by powerboaters) which is fine. There is also VMG upwind/downwind.  The Tactics instruments and documentation use different acronyms to make it clear to sailors.

canne, 2019-06-09:
Dear both, I have one rule, never fight about the meaning of the words. I remember this having been endlessly discussed already so that tom_r put in the ini-file a possibility for purists to change the name to whatever they want.

I am interested in implementing the intended function. I was wondering why the two instruments are in Dashboard but without a defined function: This mod allows the average cruiser to set a waypoint. If he happens to have the two Dashboard VMG instruments on, they should show the same value than the VMG-value in the bearing window, right?

Performance oriented sailors, like tom_r sails in racing mode. If he so likes, he puts on the VMG instruments: they need to show the VMG data from his NKE block which is sending RMB-sentences. He understands the meaning of those.

When I a make an off-shore passage, I can use the VMG instrument - if I so like - to show the VMG to the next waypoint (and see the "other VMG" from Tactics - if I so like).

People who are interested in stuff like VMG usually know what they want. There is no reason to make decision for them, just explain them how it works. Tactics documentation is a  good example of such an explanation.

transmitterdan, 2019-06-09:
I made a push to my git repo for anybody that wants to test.  Dashboard now uses RMB to drive the VMG instrument.  I will not worry about VTG.

rgleason, 2019-07-05:
> When I a make an off-shore passage, I can use the VMG instrument - if I so like - to show the VMG to the next waypoint (and see the "other VMG" from Tactics - if I so like).

I think this is ok.   I should try TDan's repos, did not see that earlier.

egilkv, 2019-08-03:
I am working on a true sailors/tactics VMG plugin which can work without additional instrumentation because it will try to figure out true wind direction, or you can set it manually. This is actually much more useful than one might think. An error in assumed wind direction does not matter much, and is is very useful, especially on a fast multihull. I will get back to this in a bit.

Anyway, it would be really good to have an agreement on what the right terms for OpenCPN are. Wikipedia https://en.wikipedia.org/wiki/Velocity_made_good uses VMG for the sailors version, VMC for the thing that GPS boxes and motorboats use. Not fighting windmills, perhaps one might use VMGwaypoint and VMGwind or some shorter version thereof?

rgleason, 2019-08-03:
We have tilted at these windmills and it is not productive.  Powerboaters use VMG(to mark) so we are using CMG (course made good upwind and downwind) in Tactics_pi which is what Tom_r decided to do and what B&G/Navionics uses.  IMHO this is settled, we are not going to change for Wikipedia. Perhaps Wikipedia should change?

egilkv, 2019-08-04:
@rgleason I do not really wish to be difficult. but the term **Course** Made Good does not make much sense to me. Based on your current course, what you want to optimize and measure is the vector component in the windward (or downwind) direction per time spent, i.e. velocity.

canne, 2019-08-04:
Hi @egilkv , a friiendly advice from a modest and volunteer plug-in programmer.  The main issue is that _**O**_ itself, in its main code and documentation is using "VMG" in its own bearing window which is part of its routing functions. Furthermore, it puts that value in a RMC sentence and sends it out and to all plugins. For the plugin programmer, that's a reserved keyword and a function name you cannot use, or if you use, you need to develop a strategy:

In tactics_pi the issue, after the unfruitful windmill fighting @rgleason was referring to, the issue was resolved by its clever programmer and sailor who, like me, understands perfectly what you want simply by explaining with [clear graphs](https://github.com/tom-r/tactics_pi/blob/master/tactics_pi.pdf) what he is producing in terms of functions. Then, not stupidly, he put the actual strings to show in the ini-file:
```
CMGSynonym=CMG
VMGSynonym=VMG
```
This way, anybody who does not agree (and there will be many) with your interpretation for the correct word to use (you have not yet thought about different language versions!) can change the provided controversial function name to whatever they want!

Windmill's arms came to a grinding halt and the programmer could concentrate to the main business: provide reliable and clearly defined functions to sailors, whatever is the name in their mind for those functions.

rgleason, 2019-08-04:
Petri, thanks I did not know about those settings!

rgleason, 2019-08-25:
BTW I would very much like to see the ability to name VMG  as VMGwpt and then have VMGwind. Maybe someday.

For the time being I believe the following improvements have been made:
1. TDan fixed showing of vmg?
2. Petri's DashT plugin takes dashboard to a new level of testing and accuracy.

Should this be closed now?

rgleason, 2019-11-14:
I am closing this now.

closed 2019-11-14 by rgleason