OpenCPN/OpenCPN #1423: Changing window size broken at least on macOS

nohal, 2019-06-08:
By resizing the window on macOS the application can get to different broken states with chart canvases not filling the window properly.
[image]
[image]
[image]

rgleason, 2019-06-09:
Windows 10 with Opencpn5.0.0+20c9a05 seems to resize fine.

nohal, 2019-06-13:
https://github.com/wxWidgets/wxWidgets/pull/1354

juanman2, 2019-06-14:
I have 5.0.0 + 9065270 on 10.12.6 (Sierra) , and I can't reproduce the resize issue.

caesar, 2019-06-18:
If this is the same issue I'm seeing, it's only with OpenGL.
See also [Flyspray 2563](https://opencpn.org/flyspray/index.php?do=details&task_id=2563).

nohal, 2022-05-04:
Fixed