OpenCPN/OpenCPN #1424: Buffer overflow and segfault loading some CM93 files

yitzikc, 2019-06-09:
Description
-----------
Certain CM93 files, like the two attached, cause an overflow of the _Cell_Info_Block::pvector_record_block_top_ buffer in _read_vector_record_table()_, leading to a segmentation fault. The size given in the header's _n_vector_record_points_ is apparently insufficient.

Environment
------------
I've reproduced the issue on 3 separate OS/X MacBooks using OpenCPN 5.0.0 and also 4.8.6. The person who provided me the files had been using them successfully on his Mac with some earlier version of OpenCPN, possibly 4.2.4, but I haven't verified that.

Reproduction
-------------
I've forked the project and created a branch demonstrating the problem in https://github.com/yitzikc/OpenCPN/tree/cm93-demonstrate-overrrun .
This should allow the problem to be recreated on the attached files. I can provide a larger subset of the CM93 files I'm using if this helps.

yitzikc, 2019-06-09:
Two files demonstrating the issue

[EN93.zip](https://github.com/OpenCPN/OpenCPN/files/3269579/EN93.zip)

rgleason, 2021-04-23:
This seems well documented, but is it a wontfix?

nohal, 2021-04-24:
What CM93 chartset are these problematic files from? Size wise they seem to be the same as the January 2011 set, but the content is quite different....

bdbcat, 2025-03-13:
Closed, OBE

closed 2025-03-13 by bdbcat