OpenCPN/OpenCPN #1425: Dashboard_pi  Vertical height gets frozen when an instrument is deleted

rgleason, 2019-06-10:
I found a new bug which upsets the dashboards when you delete an instrument, the vertical height sizing gets frozen. Then when you add that instrument back to the dashboard the vertical height unfreezes and gets sized properly. I'll try to attach a couple of screenshots.

See[ Post #552](http://www.cruisersforum.com/forums/f134/tactics-plugin-166909.html#post2905026)

Confirmed see [Post #557](http://www.cruisersforum.com/forums/f134/tactics-plugin-166909.html#post2905351)

canne, 2019-06-10:
Can confirm on O v5.0.0 Linux (Lubuntu 16.04 LTS) Dashboard - the window remains the same size. In addition to the above description, the entire window gets grey! Restarting the app and Dashboard brings the instrument window back, now with the right size and with the instruments visible.

rgleason, 2019-07-05:
I believe Canne has fixed this bug in Dashboard-Tactics_pi

canne, 2019-07-07:
Yes, but perhaps not the way it should've been done. Maybe some wxWidgets expert (which I certainly am not, just stubborn) would fix it easier. I gave up and like the instruments are always rebuilt from scratch, I applied the same brute force method and replaced their window panes entirely at the instrument removal / adding. This has some side effects, of course, like on the preferences dialog need to have the OpenCPN as the parent window, while in Dashboard it is the instrument window on which the right click was done. All is is inside the #ifdef #else conditional compilation structure. so easily spotted.

rgleason, 2021-04-10:
Is this fixed in standard Dashboard now?  I have tried to make this happen when deleting an instrument and vertical height adjustment is no longer frozen.  I will close if someone else can confirm.

Hakansv, 2021-04-10:
For me on Win10 the issue is still valid

rgleason, 2021-04-10:
Perhaps I am not doing it right to get the problem. I'll try again..

rgleason, 2023-01-16:
Now Dashboard seems to be totally fixed and not resizable.
[image]

rgleason, 2023-02-04:
I tried this with the current master build (yesterday) and it still happens.

rgleason, 2023-02-27:
I have found that this condition has improved with Hakan's work.. Closing.

closed 2023-02-27 by rgleason