OpenCPN/OpenCPN #1426: User Manual window cannot be closed in full-screen mode CPN

bc38, 2019-06-10:
The top bar with a button to close the window is hidden under the top of the screen

CarCode, 2019-06-10:
Very interesting. It's time to bury CPN now when people write such meaningful issues? They even don't know which OS they use?

bc38, 2019-06-10:
Windows10

rgleason, 2019-06-10:
bc38 Make this a new thread in the forum and post a screenshot.
Also please close this and it can be reopened later if it is indeed a problem.
Thanks

bc38, 2019-06-10:
> bc38 Создать новую тему на форуме и опубликовать скриншот. 
> Также закройте это, и оно может быть открыто позже, если это действительно проблема. 
> Спасибо

Video where you can see this inconvenience!
https://www.dropbox.com/s/2t2ngxneup01pyk/htmlwindow.mp4?dl=0

rgleason, 2019-06-10:
Baikal, I have Windows 10 and this is not a problem.  It is a separate window with the X in the upper right, if that is not the case for you use  Close   Alt+F4

bc38, 2019-06-10:
I CPN starts and runs in full-screen mode F11. Alt + F4 is impossible because the tablet.
This is an obvious problem that you need to fix so that the window will be fixed on the edges of the screen (as with the command to expand to full screen) and not floating. Or make the close button next to the back button.

rgleason, 2021-04-14:
@ bc38 Is this still an issue?

bdbcat, 2021-04-29:
Closing, inactive

closed 2021-04-29 by bdbcat