OpenCPN/OpenCPN #1435: TCP hangup

kapitan-k, 2019-07-04:
Using Linux Mint 19.1 and OpenCPN version 5.0.0 OpenCPN hangs up when I connect via TCP client (in golang or telnet, does not matter) ... port does not matter neither.
CPU usage for OpenCPN moves then to 100% and does not respond until I disconnect the client. After disconnection and further click on the chart OpenCPN crashes.
Log shows nothing special.
Any ideas?

kapitan-k, 2019-07-04:
BTW: TCP client receives messages happily.

bdbcat, 2019-07-04:
More information on your attempted configuration, please?

kapitan-k, 2019-07-04:
Protocol TCP, Address: 0.0.0.0 DataPort: 10110 but happens on every DataPort I used until now, no UserComment, Priority does not matter.
Output on this port is checked but hangs up also if it is not checked. Transmit all sentences.
Receive Input on this port is not checked but does not matter.

bdbcat, 2019-07-04:
Still not sure what is is that you are trying to do.  Create a TCP server?  If so, then the address should be something other than 0.0.0.0.

kapitan-k, 2019-07-04:
Oh sorry, then I do have a wrong understanding of the documentation, which states:
'If “0.0.0.0” is entered in the Address box, OpenCPN will act as a TCP server accepting a connection from a remote TCP client.'

bdbcat, 2019-07-04:
kapitan...
Sorry, I mean the client address should point to the server exactly.  Does it?

kapitan-k, 2019-07-04:
Depends on what you understand of 'which side is client and which is server'.
When I used 0.0.0.0 (thus OpenCPN is the 'server'/listener, according to the documentation), and connect on the 'client' with 127.0.0.1 I definitely get crashes of OpenCPN even whilst 'client' receives messages.
When I use 127.0.0.1 with 'OpenCPN' to connect it 'as a client' to 'a server' (listener) in golang everything works fine now.
So actually I'm happy with it now.
Anyway I strongly believe that 'a piece of software' (OpenCPN) should never crash / hang up by a 'client' connecting which happens in my case.
But I'm not sure if its worth investigating.

bdbcat, 2019-07-05:
Probably something going on here, although the server (0.0.0.0) configuration is commonly used on other linux platforms without problem.  Would be worth running under gdb, and getting a stack trace on the fault, if possible.

scadaperspective, 2019-07-05:
Ive found using O as a server in that config oV50 it becomes a very fragile thing on macOS as well, but I am still analysing the root cause(s) 

I have found rebooting after setting up oV50 as a server seems to stop the initial crash caused by a client initiating the TCP connection but about 4 hours and things get really slow and a reboot Is required of the app 

For it to act as a server is  it using GPSD?  I know I have GPSD installed on my mac Instances but don’t have the service running as per the brew install advice,  so I assumed this was something to do with it ? 

If it isn’t configured as a server the tcp connections are a lot more robust. Dashboard can be a bit sluggish in seeing the connection changes and to start parseing on not parseing sentences. 

Will grab a crash dump and upload it next time I am playing with tcp config’s 

/Ron

bdbcat, 2019-07-05:
We do not use the GPSD library.  I found it to be fragile, also.  We parse GPSD messages as received.
Interestingly, we made very superficial changes to TCP/IP code base between O4 and O5.  So I would be interested in a crash dump if possible.

kapitan-k, 2019-07-05:
Here you go with the gdb output ... I connected two times and moved the cursor a bit and with the second time disconnecting.

GDB output:
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x0000555557ce4170 in ?? ()

`
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x00007ffff573126b in ?? () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
(gdb) bt
#0  0x00007ffff573126b in ?? () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#1  0x00007ffff1781185 in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff1781550 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff1781862 in g_main_loop_run () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007fffede92a37 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#5  0x00007ffff5729b05 in wxGUIEventLoop::DoRun() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#6  0x00007ffff50dca93 in wxEventLoopBase::Run() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff50a40a6 in wxAppConsoleBase::MainLoop() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff512eae9 in wxEntry(int&, wchar_t**) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00005555556d9628 in main ()

`

kapitan-k, 2019-07-05:
Valgrind output ... see line 553:
[opencpn_valgrind.txt](https://github.com/OpenCPN/OpenCPN/files/3361101/opencpn_valgrind.txt)

scadaperspective, 2019-07-05:
And from macOS ....

/Ron

Process:               OpenCPN [22801]
Path:                  /private/var/folders/*/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.0.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenCPN [22801]
User ID:               501

Date/Time:             2019-07-05 15:52:28.782 +1000
OS Version:            Mac OS X 10.14.5 (18F132)
Report Version:        12
Anonymous UUID:        2C5872C6-0E22-3A8E-45C3-EBE5776AFDF2

Time Awake Since Boot: 330000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000071
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [22801]

VM Regions Near 0x71:
--> 
    __TEXT                 00000001036f8000-0000000103ef0000 [ 8160K] r-x/rwx SM=COW  /var/folders/*/OpenCPN.app/Contents/MacOS/OpenCPN

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libwx_baseu_net-3.1.2.0.0.dylib	0x00000001042de47a wxSocketBase::Destroy() + 10
1   org.opencpn                   	0x000000010388b341 DataStream::OnSocketEvent(wxSocketEvent&) + 2065
2   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104b0001f wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 367
3   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104b00d7a wxEvtHandler::ProcessEventLocally(wxEvent&) + 90
4   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104b00c70 wxEvtHandler::ProcessEvent(wxEvent&) + 96
5   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104b00ae3 wxEvtHandler::ProcessPendingEvents() + 243
6   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104a18717 wxAppConsoleBase::ProcessPendingEvents() + 119
7   libwx_baseu-3.1.2.0.0.dylib   	0x0000000104ae16a2 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 66
8   com.apple.CoreFoundation      	0x00007fff44c61734 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
9   com.apple.CoreFoundation      	0x00007fff44c61669 __CFRunLoopDoObservers + 451
10  com.apple.CoreFoundation      	0x00007fff44c041fe __CFRunLoopRun + 1143
11  com.apple.CoreFoundation      	0x00007fff44c03b35 CFRunLoopRunSpecific + 459
12  com.apple.HIToolbox           	0x00007fff43ee296b RunCurrentEventLoopInMode + 292
13  com.apple.HIToolbox           	0x00007fff43ee25ad ReceiveNextEventCommon + 355
14  com.apple.HIToolbox           	0x00007fff43ee2436 _BlockUntilNextEventMatchingListInModeWithFilter + 64
15  com.apple.AppKit              	0x00007fff4227c987 _DPSNextEvent + 965
16  com.apple.AppKit              	0x00007fff4227b71f -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
17  com.apple.AppKit              	0x00007fff4227583c -[NSApplication run] + 699
18  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x00000001045146ae wxGUIEventLoop::OSXDoRun() + 174
19  libwx_baseu-3.1.2.0.0.dylib   	0x0000000104ae1cfc wxCFEventLoop::DoRun() + 44
20  libwx_baseu-3.1.2.0.0.dylib   	0x0000000104a409d5 wxEventLoopBase::Run() + 85
21  libwx_baseu-3.1.2.0.0.dylib   	0x0000000104a17f64 wxAppConsoleBase::MainLoop() + 116
22  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x00000001044b2baa wxApp::OnRun() + 26
23  libwx_baseu-3.1.2.0.0.dylib   	0x0000000104a6ffe8 wxEntry(int&, wchar_t**) + 56
24  org.opencpn                   	0x000000010379c474 main + 20
25  libdyld.dylib                 	0x00007fff70bcc3d5 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff70dbf3f0 start_wqthread + 0

Thread 2:
[... 435 lines cut ...]
Memory Module: DIMM4, 16 GB, DDR3 ECC, 1866 MHz, 0x80AD, 0x484D54343247523744465234432D52442020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x135), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.12f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en2
PCI Card: AMD FirePro D700, Display Controller, Slot-2
PCI Card: AMD FirePro D700, Display Controller, Slot-1
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: eLicenser
USB Device: USB2.0 Hub
USB Device: LaCie Hard Drive USB
USB Device: USB2.0 Hub
USB Device: E-MU 0404 | USB
USB Device: Microsoft® Nano Transceiver v1.0
USB Device: Hub
USB Device: MG-XU
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2

CarCode, 2019-07-05:
scadaperspective

Why you have OpenCPN here:  /private/var/folders/*/OpenCPN.app
and not in /Applications/ as usual in macOS ?

scadaperspective, 2019-07-05:
Because it is a Mac Pro

CarCode, 2019-07-06:
> Because it is a Mac Pro

Even a Mac Pro runs macOS. So you are doing something wrong.
As seen you are running Mojave 10.14.5. Try to use it as Apple recommends.

scadaperspective, 2019-07-07:
Apologies to everyone else 

CarCode,

I have resisted replying to you on other occasions and mediums however I decided to try to give you one last opportunity to pull your head in as we say in Australia. 

Please do pull your head in go sailing and stop bothering nice people.

This reply is the last time I will respond to you so your 15 minutes with me is now over.

As far as I am concerned Car Code your comment sadly is as frequently and usually the case nothing to do with what is being posted. 

It add's no benefit to me or the conversation at hand and your comments directed towards me are completely unsolicited. 

Please don't communicate with me unless I solicit such a response, unless you use / provide your real first name and not some pseudonym or at the very least you are actually being helpful. 

I have not had you introduced to me as being part of the actual support team or development team, so why are you commenting at all here unless it is to report or convey an end user issue?

 Do you work for Apple computer's Inc USA ? No, I suspect not as you would know better otherwise.

The Machine the dump was captured from IS an Apple Mac Pro business workstation build by Apple computers Inc, it is a standard end user build and not a special features build, It was purchased from them by me as a regular apple business customer, all software applications including operating system was/is installed by them and it was shipped to Apple Australia (Sydney) for final testing before being shipped to me and remained under a controlled chain of custody now residing with me.
 
It comes with the configuration I purchased from them. I have made no alterations to the machine apart from installing software and hardware approved for use by Apple computers Inc. I am using this equipment correctly, in fact actually my usage is expressly approved by Apple computers Inc as being fair and reasonable use of the product and in full accordance with their terms and conditions of sale in Australia. So you're wrong.

The problems with wxWidgets and for Apple specifically with Xcode frameworks I have been experiencing and are now resolved are external to and un related to the issues associated with wxWidgets 3.1.2 or 3.1.3 for that matter, are more general OpenGL compatibility issues related to the support library, issues that I and others are genuinely experiencing and trying to assist in resolving, people including Mike are reporting is the actual topic being discussed, when you, as usual, rudely interjected. 

Please don't do it any more, and more importantly for you please don't do it towards me specifically as I will take affirmative actions to have this pseudonym you are using be blocked here and on the cruisers forum if you continue.

Ron Southworth

rgleason, 2021-04-23:
@kapitan-k  @bdbcat   Has this been resolved? Or should it stay open?  Original poster needs to close it.

kapitan-k, 2021-04-24:
No resolution. Not important for me, no further issues / comments of others. 
Thus, I close it.

closed 2021-04-24 by kapitan-k