OpenCPN/OpenCPN #1437: ENC chartcorrection files

RooieDirk, 2019-07-06:
I have some ENC charts where the chart corrections are in a submap inside the chart map. 
[image]
However O does not find the correction and only the plain standard ENC is displayed. So if here is a sub-map inside a chart-map O should also have a look inside there.

bcn2, 2019-07-07:
I'm not sure if this structure conforms with the S-57 standard.
ENC_ROOT is mandatory and within the same directory the CATALOG.031 file.
ENC_ROOT_Ed1 is not a valid name to designate a root directory.
The updates would need their own ENC_ROOT folder and CATALOG.031 file
(OpenCPN does ignore the CATALOG.031 file)
Unfortunately the standard does not force a specific structure and each HO is inventing its proper one.

RooieDirk, 2019-07-07:
The files that you get from the Dutch HO for OeSenc, do they have the same structure? Or did somebody without to much knowledge mess around with the files before I got them?

bcn2, 2019-07-07:
The NL sets are delivered by UKHO. As an example for their file structure of the delivery sets, the NL charts for the Caribbean. You can take the US S-57 sets as another valid example.
[image]

closed 2019-12-05 by RooieDirk