OpenCPN/OpenCPN #1438: Android build error with WxWidgets

CptMonkeyFist, 2019-07-08:
I set up a build environment to build OpenCPN for android using the "android" branch. I even set up the a "dsr" user to ensure no path conflicts. I downloaded the android sdk tools, android ndk, and Qt 5.2.1 as per the readme in the buildandroid folder, steps A, B, and C. I downloaded the modified WxWidgets repository as specified in step D. I configured it as instructed in the document, but when I went to build it I got the following error:
```
In file included from ../include/wx/wxprec.h:42:
In file included from ../include/wx/wx.h:24:
In file included from ../include/wx/event.h:21:
In file included from ../include/wx/cursor.h:63:
../include/wx/qt/cursor.h:15:10: fatal error: 'QtGui/QCursor' file not found
#include <QtGui/QCursor>
         ^~~~~~~~~~~~~~~
33 warnings and 1 error generated.
make: *** [.pch/wxprec_corelib/wx/wxprec.h.gch] Error 1

```
I checked the include directory for my Qt installation and QCursor.h and QCursor.c exist in the QtGui directory. 

While I realize that this is more of a WxWidgets issue, the "stable" version of WxWidgets used by the instructions is maintained by Dave Register, the person selling the android port and there is no clear way of raising a github issue with that version of WxWidgets. Any help would be appreciated.

I have also tried this with Qt 5.3.2 and got the same results.

bdbcat, 2019-07-08:
The github repo for branch Android, and the associated wxWidgets port, are not in great shape at the moment.  This is on my list for a summer project, to bring the repos and build process back up to current state.

But meanwhile, we can muddle along.
The error you found is probably due to missing the PKG_CONFIG_PATH environment variable in your build shell.  I usually configure the wxWidgets build like this:

../configure --with-qt --build=x86_64-unknown-linux-gnu --host=arm-linux-androideabi --enable-compat28 --disable-shared --disable-arttango --enable-image --disable-dragimage --with-libtiff=no --disable-baseevtloop --disable-xrc --disable-cmdline --disable-miniframe --disable-mdi --disable-stc --disable-ribbon --disable-propgrid --disable-timepick --disable-datepick --enable-xlocale --enable-intl --with-opengl --enable-selectloop PKG_CONFIG_PATH=/home/dsr/Qt5.5.0/5.5/android_armv7/lib/pkgconfig

You may need to adjust the PKG_CONFIG_PATH parameter to match your Qt installation.

Dave

CptMonkeyFist, 2019-07-09:
I figured there was something wrong with PKG_CONFIG_PATH when I saw the error, which is why I double checked it before submitting this issue. It seems as though exporting the PKG_CONFIG_PATH variable does not work as expected. Your modified configure parameters seemed to do the trick for solving this issue, but another has arisen:
```
../src/common/intl.cpp:797:25: error: use of undeclared identifier 'nl_langinfo'
    const char *alang = nl_langinfo(CODESET);
                        ^
../src/common/intl.cpp:1650:24: error: use of undeclared identifier
      'nl_langinfo'
    const wxString fmt(nl_langinfo(items[nlidx]));

```

I'll admit I wouldn't mind just paying for your android port, but it doesn't exist on the Amazon Fire app store, if you're lucky you could just submit the same APK to that store as well. In the meantime, yeah... a cleaned up set of build instructions would certainly be appreciated. I'm concerned that newer code will not work for the earlier instructions meant for older versions of the project and its dependencies.

I'll look into this new issue as best I can on my own. If you would prefer I create a separate issue for each problem that I cannot solve on my own, then let me know.

CptMonkeyFist, 2019-07-09:
I did some Googling and according to one person, nl_langinfo() was introduced to Android at API 26. The current generation of Amazon Fire Tablets use a modified version of Android 7.1 which uses API 25. I'll have to see if there is some way around this, otherwise I may be out of luck.

CptMonkeyFist, 2019-07-09:
At this point I don't think it is fair to leave this issue open. The android port was clearly meant for devices with newer versions of android installed. I may just buy an android tablet for this purpose since it costs about as much as buying a large enough 5V screen for an embedded linux board like the raspberry pi. I appreciate your help @bdbcat.

closed 2019-07-09 by CptMonkeyFist