OpenCPN/OpenCPN #1439: Question: how are static/non-scaling features rendered?

jaredrada, 2019-07-11:
I am in the process of building my own mapping system and I'm struggling conceptually with how to handle certain rendering aspects. Was hoping to learn from this project!

On my map I have roads- roads get smaller and larger as the user zooms in or out. But I also have road labels (like street names). My problem is I want some objects (like street names or icons) to be exempt from scaling- such that as a user zooms in/out the road name label remains the same size while the road shape scales.

How does OpenCPN handle this? I see two options: 1. as the user zooms into the map, the road names can be scaled down such that they remain a constant size (and vice verse for zooming out). OR option 2. as the user zooms into the map, create a "static size" layer and calculate the relative position of "static size" icons. 

Any insight into the OpenCPN algo for this is much appreciated. Thanks for any/all help.

CarCode, 2019-07-12:
Do you really think such a question is correctly placed into the issues chapter of this project?
Why not use the forum at http://www.cruisersforum.com/forums/f134/ ???

jaredrada, 2019-07-12:
Absolutely. Since it’s algorithm related and not general “how to” it belongs in the repo. I would tag it with question like most other open source repos. I’ve asked plenty of other questions here, feel free to sort by author.

CarCode, 2019-07-15:
>  I’ve asked plenty of other questions here, feel free to sort by author.

And how many answers you got here? I don't see any.

rgleason, 2019-10-23:
I'm with Carcode. Please put your post in a new thread here
http://www.cruisersforum.com/forums/f134/ and you will soon have an answer.
Also please close this post. Thank you.

closed 2019-12-31 by jaredrada