OpenCPN/OpenCPN #1441: Sound issue on all platforms

transmitterdan, 2019-07-22:
In ```void options::OnButtonTestSound(wxCommandEvent& event)``` we initialize ptest_sound with a pointer obtained from ```new``` but never delete the pointer.  Each time the user tests the AIS alert sound after the first time a small leak is created.

leamas, 2019-07-22:
Right. Here are also more problems, one is updating the command string unconditionally although it is not used in any default build configuration. It should really be guarded by `if ((bool) dynamic_cast<SystemCmdSound*>(ptest_sound) )`

Also, my gut feeling is that the static variable ptest_sound could and should be avoided. But, my Yanmar is definitely a more pressing issue.

Seems that the culprit is [363a754b21] when it comes to the memory leak and the static, the command string stuff was there from the beginning.

Also note that `onTestSoundFinished` does not do anything, and is most likely optimized to {} .

On Android there is a possibility to issue a new call while the sound is still playing. It will then be silently dropped. Guess we could live with that.

leamas, 2019-08-26:
Attaching untested patch which should fix the leak. If anyone could test it and make a PR it would be fine. Or, I could make a PR if someone tests it but don't want to make the PR

[0001-sound-Fix-memory-leak-1441.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3541051/0001-sound-Fix-memory-leak-1441.patch.gz)

leamas, 2019-08-26:
And here is a revised form of the patch, keeping things in a single function.

[0001-sound-Fix-memory-leak-1441.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3541209/0001-sound-Fix-memory-leak-1441.patch.gz)

Looks better to me. One open issue is that this logic assumes that all ARM devices are "slow" in this context. This is probably not true and should ideally be probed by cmake.

leamas, 2019-09-22:
hm... I have found some time to test it myself and no, it's not ok. Back later.

PR #1466 (2019-12-07): Sound: Fix memory leak + portaudio close issue (#1441)
Closes: #1441
commit b98721a5 (2019-12-07): options: Fix sound test memory leak (#1441).

The fix uses asynchronous operation on all platforms to get
a uniform behaviour.

Closes: #1441
closed 2019-12-07 by bdbcat