OpenCPN/OpenCPN #1442: Build Plugin

jishatomc, 2019-07-24:
@ rgleason

As you said I went through Developer Guide, and installed OpenCPN, I am able to compile and execute. I have downloaded the draw plug-in and installed, unfortunately it is not satisfying my requirement to draw some particular shape and move with a certain speed.

Can you help me to develop a plug in for this purpose.
If you have some tutorial for developing a please share.

Thanks in advance,
Jisha.

rgleason, 2019-07-24:
Jisha,
I think you will need to review the code in the github repository of Jon Gough  Ocpn_Draw author.
https://github.com/jongough/ocpn_draw_pi and add some of the functionaliity you are looking for.
I suggest you fork his code and the work locally.

Make a new branch for your work.
There are many examples of objects moving with the boat or not, PIL, EBL.
There are many examples of creating objects that draw on the screen, like Boundaries.
Then if you build something that is useful for navigation, then perhaps Jon Gough will accept and merge your PR>

I thiink you should be using the OpenCPN Forum here 
http://www.cruisersforum.com/forums/f134/  
for these kinds of questions.   Create your own thread, with appropriate title, and then describe your goal and post your questions. We will try to help.

rgleason, 2019-07-24:
Now it is time for you to close this "Issue" because it is not focused on an Issue with Opencpn itself, it is an Issue you have. Thank you.

rgleason, 2019-08-25:
Can someone please close this?

rgleason, 2019-11-25:
This should be closed, I  believe.

rgleason, 2019-12-14:
You can create a boundary and move it using Ocpn-Draw.  
jishatomic, it is time to close this issue!

closed 2021-04-13 by bdbcat