OpenCPN/OpenCPN #1443: RFE: Better plugin versioning

leamas, 2019-07-25:
Currently, the plugin API  exports two symbols for major and minor plugin version. This works as a basic check which is displayed in the Options | Plugins UI.

However, when implementing a plugin loader it becomes necessary to judge if a downloadable plugin represents an upgrade, downgrade or a reinstall of the existing version. For this, a more fine-grained versioning is required.

Looking around in the opencpn community it seems like a version like *major.minor.patch.build* sort of codifies what's already used. Of these, the build is an arbitrary text string and the rest integers for example *3.2.0.1-deadbee*.

I suggest that we add a version property to the plugin interface.    For plugins not supporting this new feature we use a version constructed from current major and minor, like *"3.2.0"* in previous example

rgleason, 2019-07-26:
Thanks Leamas, that will help users know if they have a correct plugin for major upgrades involving wxwidgets and will keep the plugin targets separate during transition.

leamas, 2019-07-26:
The Yanmar is dead, broken beyond repair.

rgleason, 2019-07-26:
Darn. Very Sorry.  We had that with Westerbeke in 2013 at the start of the season. Mechanic replaced in 2 weeks and saved our season.

rgleason, 2019-08-19:
The need for this was recently driven home to me by Petri, when we tried to get version major.minor.patch  to show in as  the Preferences header as  "Tactics 1.0.10", and a better Plugin API would definitely help.

See the screenshot and comments here
https://github.com/rgleason/tactics_pi/commit/70c75e08982a0ea75b07d64dc48bd06c9538b758

Also here Petri's solutions down towards the bottom.
https://github.com/tom-r/tactics_pi/issues/38

This type of thing does not lead to interface consistency.
I wonder if there are other values that should be in the Plugin API?

rgleason, 2019-08-19:
I think we need  in the Plugin's main  *_pi .h file 
under 
//    The required PlugIn Methods
`     int GetPlugInVersionPatch();
`
and in the main *_pi.cpp file

```
int tactics_pi::GetPlugInVersionPatch()
{
    return PLUGIN_VERSION_PATCH;
}
```
OpenCPN currently knows when to spit a plugin out due to incompatible API using GetAPI....
but is there anything else that the main program needs to know?

leamas, 2019-08-20:
>  think we need in the Plugin's main *_pi .h file

IMHO, it should be

    int GetPlugInVersionPatch();
    const char* GetPlugInVersionBuild();
    
The build field could be used for things like date and/or git hash or so. The important thing is that it sorts ok. One well-tested variant is date-hash like *20190818_deedbee*. But from the API perspective it's just a free-format string.

> but is there anything else that the main program needs to know?

Frankly, no. It needs to test the API version anyway and handle the missing functionality using defaults.

commit db9f0a64 (2019-12-07): Handle Pango >=1.44 Harfbuz dep (Closes: #1443).

Recent pango libraries (1.44+) has a harfbuzz dependency.  Add a
new FindPango and FindHarfBuzz cmake finders and use them as
appropriate.
PR #1460 (2019-12-14): Version: Extend plugin version info API  + display extended version info
Add some new attributes to the plugin API to establish a full plugin semantic version. Displays the complete plugin version and also makes the plugin API version visible in both Options | Plugins and About.

The PR closes #1443 and also handles IMHO still valid aspects of #1445.

This PR is not fully coordinated with #1457. Depending on which is merged first there is some rebasing to do.
closed 2019-12-07 by bdbcat