OpenCPN/OpenCPN #1445: Plugin API improvement: Show version major, minor, patch

rgleason, 2019-08-01:
It would be helpful to know the exact Plugin Version (major,minor,patch) in Options>Plugins listing and in the Preferences window of every plugin.

For example, Ocpn_Draw in the About page has:
Ocpn_Draw in the About page shows
Plugin Major Minor Patch.
Plugin API number
OpenCPN API number   (maybe version would be better?)
Draw API number
Date

I think it will help Programmers, Troubleshooters, Beta Testers and Users.

leamas, 2019-09-19:
Now, despite the heading this bug is more about the GUI than the API. The actual description here describes the GUI, and the API changes are covered in #1443. 

I share your view that major, minor, patch, pre-release tag and build info should be displayed in  Options | Plugins. The date could go into the build info but this is eventually down to plugin developers. 

However, I hesitate about the API number info. This is nitty-gritty details that would bog down the panel with info which is irrelevant for most users (after all, the API level is not a major problem since the plugin is successfully loaded...)

Perhaps we could have some kind of Details button besides preferences covering this stuff and perhaps more. But then again,  this seems to be little outside the scope of this bug. It could covered by another bug focused on "Plugin details" or so. It's actually not that related to the version info.

Bottom line: I tend to think we should close this bug when we can display complete version info, possibly then filing new bugs as required.

rgleason, 2019-09-19:
I believe Bdbcat said the API should be extended and it is covered by #1443 

I believe it is important to also display Ocpn_min_version  (OpenCPN Minimum Version) Like "ov50"
Should I changed the title to  version, major, minor, patch and ocpn_min_version?
- Is that what you are calling "build".   

I think it is very important to name it specifically for plugin authors understanding, "build" means something different, also we need it because users don't know about wxWidgets and assume that plugins are always forward compatible.  Also this is necessary while transitioning to a new wxWidgets.

You want to close this. I don't think the majority of plugins have changed to this yet.

leamas, 2019-09-20:
> I believe Bdbcat said the API should be extended and it is covered by #1443

Right

> I believe it is important to also display Ocpn_min_version (OpenCPN Minimum Version)
> Like "ov50 Should I change the title to version, major, minor, patch and ocpn_min_version?

That is problematic for two reasons. One is that this is not available in the plugin API, what's there is the plugin API version. It *is* possible to construct a mapping from plugin API versions to Opencpn versions, but I'm not really inclined to do it.

Because, secondly, the important usecase here is if the plugin cannot be used with the Opencpn version being used. And in this case it makes no sense since the plugin isn't loaded at all, nothing is displayed.

> Is that what you are calling "build".

Not me. This is the standard being used nowadays for versions info which is called Semantic Versioning, https://semver.org.

> You want to close this. I don't think the majority of plugins have changed to this yet.

This is just not how things work.  This tracker is about opencpn, and work we need to in it. Work which should be done in the plugins should be filed in bugs against those plugins.

So, IMHO we should close this bug when the necessary  tools for the plugins are in place. No more, and no less.

rgleason, 2019-09-20:
Alec,

You are focused on the Plugin Installer, there is the existing installation situation to consider. With no timetable, since installation is manual for windows at least, there is a good reason for "ov50" and oocpn_min_version.

I am aware of semver.org and have posted it several times in the forum.

I am going to file that in 41 plugins?  Forget it.

rgleason, 2019-09-20:
If someone can improve and clarify  the statement in a new Issue. I will close this.

rgleason, 2019-09-20:
`Because, secondly, the important usecase is if the plugin cannot be used with the Opencpn version being used. And in this case it makes no sense since the plugin isn't loaded at all, nothing is displayed.`

Yes, and the new user thinks they installed a plugin, but it is not loaded, so they write in thee forum and are told to go look at the opencpn.log to see what happened.  Since they are installing manually, why not give them a visual clue as to what versions of Opencpn it will work with?

Now with your new Plugiin Installer, the plugins are all listed and prequalified and ready to be installed. How nice!!!  ...and we don't have to worry about that issue.

rgleason, 2019-09-20:
Since the Plugin API does not handle ocpn_min_version, why not just have this value set in CMake by the plugin authors for the time being.

leamas, 2019-09-20:
Well, it seems this issue  then is about the plugins and not primarely Opencpn. As such, I cannot see any patches to Opencpn which can resolve the issue.

How are independent plugin devs supposed to discover this bug?

leamas, 2019-09-20:
> Since the Plugin API does not handle ocpn_min_version,

But it does, by refusing to load plugins with a plugin API mismatch (basically, that the plugin requires a newer version of Opencpn). It's just that this is not expressed in OpenCPN versions but plugin API versions. This is IMHO a  technically sound decision.

We could of course log errors when trying to load incompatiible plugins  in a GUI message box. But then again, where does this bug start and end?

rgleason, 2019-09-20:
I give up

closed 2019-09-20 by rgleason

leamas, 2019-09-21:
Hm... With respect to filing bugs against the plugins on the opencpn bugtracker this probably the right decision.  However, if we keep our eyes  on the ball (opencpn):

> Yes, and the new user thinks they installed a plugin, but it is not loaded, so they write in the 
> forum and are told to go look at the opencpn.log to see what happened. Since they are 
> installing manually, why not give them a visual clue as to what versions of Opencpn it 
> will work with

This an important finding. Plugin compatibility is about the plugin API version, which is different from the opencpn version. A plugin is compatible if it's built for a plugin API version  <= current API version in opencpn. 

However, there is no visible user information whatsoever  on this. The About page only displays the version, not the plugin API version.  And, as you said elsewhere, the  Options | Plugins page does not display the plugin API version either. This is IMHO things which should be filed and fixed.

If this problem is big enough to motivate it it would also be possible to construct a list on "unloaded plugin files". This would basically be a list of filenames together with the reason it's not loaded such as not activated, linkage errors, wrong API version, already loaded, etc. The list could be part of Options | Plugins under a button like "Unused plugin files" or so.

Bottom line: I see at least two good issues which could be filed against opencpn based on this discussion. That said, I still think plugin issues should be filed in relevant plugin trackers.

rgleason, 2019-09-22:
Those are good ideas. Thank you Leamas.
I went back to look at the developer manual because the[ Plugin API page ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api:plugin_api_versions)gets added to at each change of the Plugin API.  This is where we have a table of Plugin API to Opencpn Version

rgleason, 2019-09-22:
Alec,
While testing Tom_r recent tactics_pi I discovered that there is a message to the user when a plugin is not compatible!  An improvement might be to state the Plugin API and the Opencpn required API, but this isn't the problem I thought it was.  Sorry.

[image]