OpenCPN/OpenCPN #1447: Problem in (Georef) ll_gc_ll_reverse(...)

RooieDirk, 2019-08-22:
In this [http://www.cruisersforum.com/forums/f134/add-second-gps-object-to-opencpn-221605.html](url) forum thread a problem showed up. After some investigating I found that the function ll_gc_ll_reverse(..) sometimes give's a NAN for Distance and Bearing. (While if the distance = zero all bearings upto 360 are true, and the distance should always be a number)

While searching for the problem I also did compare above function with the DistanceBearingMercator(..) function. On short distances the outcome should be more or less the same. `DistanceBearingMercator( 44.391417, 8.917939, 44.391417, 8.917937, 90.000000, 2.323207E-01 )
ll_gc_ll_reverse       ( 44.391417, 8.917939, 44.391417, 8.917937, 270.000001, 7.257515E-05 )

DistanceBearingMercator( 44.391417, 8.917936, 44.391418, 8.917934, 97.225663, 2.862195E-01 )
ll_gc_ll_reverse       ( 44.391417, 8.917936, 44.391418, 8.917934, 277.225664, 7.257318E-05 )

DistanceBearingMercator( 44.391418, 8.917934, 44.391417, 8.917934, 0.000000, 3.600000E-02 )
ll_gc_ll_reverse       ( 44.391418, 8.917934, 44.391417, 8.917934, -nan, -NAN )

DistanceBearingMercator( 44.391417, 8.917934, 44.391417, 8.917934, 324.358096, 2.214908E-02 )
ll_gc_ll_reverse       ( 44.391417, 8.917934, 44.391417, 8.917934, -nan, -NAN )

DistanceBearingMercator( 44.391417, 8.917934, 44.391417, 8.917934, 55.111383, 6.293894E-02 )
ll_gc_ll_reverse       ( 44.391417, 8.917934, 44.391417, 8.917934, -nan, -NAN )

DistanceBearingMercator( 44.391417, 8.917934, 44.391417, 8.917932, 90.000000, 1.806950E-01 )
ll_gc_ll_reverse       ( 44.391417, 8.917934, 44.391417, 8.917932, -nan, -NAN )

DistanceBearingMercator( 44.391417, 8.917932, 44.391418, 8.917930, 106.638726, 3.771792E-01 )
ll_gc_ll_reverse       ( 44.391417, 8.917932, 44.391418, 8.917930, 286.638727, 1.026223E-04 )

DistanceBearingMercator( 44.391418, 8.917930, 44.391418, 8.917928, 90.000000, 2.581325E-01 )
ll_gc_ll_reverse       ( 44.391418, 8.917930, 44.391418, 8.917928, 270.000001, 7.257515E-05 )
`
Turns out that the bearing of DistanceBearingMercator() is 180 degrees off.
I'll investigate further.

rgleason, 2021-04-14:
Rooie, did you investigate further?  What happened?  Can this be closed?

RooieDirk, 2021-04-16:
Ah still open. Is implemented already I'll close it

closed 2021-04-16 by RooieDirk