OpenCPN/OpenCPN #1450: Black Squares on S57 OSX

grhitch, 2019-08-28:
The are surrounding a sectored light shows as black.
This can be  seen using the IHO Test Data Set 6.4.1 - Power Up and going to  32 28.7942S 060 58.8138E

The installed version is: 5.0.1+0266678
OSX Version: Mojave 10.14.5
Machine MacBook Pro Retina 13 Late 2012
Enclosed are two screen shots, one form ocpn and one from MacENC showing the same charts rendered correctly.
[image]
[image]

CarCode, 2019-08-28:
Strange, the position you mentioned is somewhere in the Indian Ocean south-east of Madagascar without any land.
[image]

bcn2, 2019-08-28:
As said above: These are test charts from the S64 ENC test set - "fake cells".
OpenCPN shows 32 25S instead of 32 28S btw... 

But should be presented correctly anyhow.
Which version of the S-63 plug-in is used? Does OpenGL on/off make any difference?

bcn2, 2019-08-28:
There is a more actual version available from o-charts.org/downloads.
Would you mind to try that one?

bcn2, 2019-08-28:
Ok, did download the charts from here:
https://www.iho.int/iho_pubs/standard/S-64/S-64Ed3_0_0/index_new_20150109.htm
and using the "power-up" set, all is fine (macOS 10.14.6, OCPN 5.0.1)
[image]

grhitch, 2019-08-28:
Thanks, I am still on 10.14.5, I will try updating as soon as the update server shows it as available for me.
Reading the release notes, I don't have much faith that this will fix it, but happy to try.
I have done check sum comparisons of all the s57 critical rendering files with a working 5.0.0 Linux installation and cannot find any differences. 
I have also tried setting DebugS57 in the ini file, but don't see anything relevant in the log file.

grhitch, 2019-08-28:
S-63 Plugin is 1.12 and have tried Open GL on and off and have seen no difference.

grhitch, 2019-08-28:
Ok, 10.14.6 has come through and as I feared, no difference.

grhitch, 2019-08-28:
I can only find the oeSENC plugin from the OpenCPN site. This is not really relevant,tThe test cells are S57 and shouldn't even require the s63 plugin> Tried disabling and also a fresh install without the plugin, both with the same result.

grhitch, 2019-08-29:
I have cleared out the directories SENC  and raster_texture_cache. One of these fixed the problem. Not really sure how as the problem existed on s63 charts as well and these don't seem to have anything stored in either of these directories.
Thanks for suggestions.
BTW the S63 plugin rports itself as 1.12, but the file that it is installed from was S63-Plugin_1.13.1404.pkg

closed 2019-08-29 by grhitch