OpenCPN/OpenCPN #1453: Pango 1.44 requires harfbuzz, cmake needs to take this into account

taruti, 2019-09-05:
Current master fails on new Arch Linux with:
```
[  8%] Building CXX object libs/wxsvg/CMakeFiles/WXSVG.dir/src/cairo/SVGCanvasTextCairo.cpp.o
cd /home/t/w/OpenCPN/build/libs/wxsvg && /usr/bin/c++  -DUSE_RENDER_CAIRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -DocpnUSE_GL -DocpnUSE_SVG -DwxUSE_WEBVIEW=0 -DwxsvgUSE_EXIF -I/usr/include/libexif -I/home/t/w/OpenCPN/libs/wxsvg/include -I/home/t/w/OpenCPN/libs/wxsvg/include/wxSVG -I/home/t/w/OpenCPN/libs/wxsvg/include/wxSVGXML -I/usr/include/cairo -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/pango-1.0 -I/usr/include/gtk-2.0 -I/usr/lib64/gtk-2.0/include -isystem /usr/lib/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0  -pthread -O3 -DNDEBUG    -DPREFIX=\"/usr/local\" -fvisibility=hidden -Wall -Wno-unused -fexceptions -rdynamic -fno-strict-aliasing -Wno-deprecated-declarations -std=gnu++11 -o CMakeFiles/WXSVG.dir/src/cairo/SVGCanvasTextCairo.cpp.o -c /home/t/w/OpenCPN/libs/wxsvg/src/cairo/SVGCanvasTextCairo.cpp
In file included from /usr/include/pango-1.0/pango/pango-font.h:25,
                 from /usr/include/pango-1.0/pango/pango-attributes.h:25,
                 from /usr/include/pango-1.0/pango/pango.h:25,
                 from /usr/include/pango-1.0/pango/pangocairo.h:25,
                 from /home/t/w/OpenCPN/libs/wxsvg/src/cairo/SVGCanvasTextCairo.cpp:22:
/usr/include/pango-1.0/pango/pango-coverage.h:28:10: fatal error: hb.h: No such file or directory
   28 | #include <hb.h>
      |          ^~~~~~
compilation terminated.
make[2]: *** [libs/wxsvg/CMakeFiles/WXSVG.dir/build.make:833: libs/wxsvg/CMakeFiles/WXSVG.dir/src/cairo/SVGCanvasTextCairo.cpp.o] Error 1
```

This is due to recent Pango depending on Harfbuzz (`-I/usr/include/harfbuzz/` fixes this).

Related issues:
https://bugzilla.redhat.com/show_bug.cgi?id=1741077
https://github.com/eiskaltdcpp/eiskaltdcpp/issues/413
https://gitlab.gnome.org/GNOME/pango/issues/387  

Cmake magic should be adjusted to get correct harfbuzz path and add that when using (recent) Pango.

leamas, 2019-09-10:
hm... Do  you know how *recent* is defined in this context? That is, from which version does pango pull in harfbuzz?

leamas, 2019-09-10:
Perhaps you have time to test this patch which is supposed to fix it?

[0001-Handle-Pango-1.44-Harfbuz-dep-1443.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3596771/0001-Handle-Pango-1.44-Harfbuz-dep-1443.patch.gz)

taruti, 2019-09-11:
Unfortunately that does not make the build work. (Deleted old build directory, created a new one, ran cmake and make). Not sure if it is relevant that the failing compile is inside wxsvg.

From cmake output:
```
-- SVG support enabled...
--  Cairo library found: /usr/lib/libcairo.so
-- Found CAIRO: /usr/include/cairo  
-- Checking for one of the modules 'pango'
-- Found Pango: /usr/lib/libpango-1.0.so (found version "1.44.6") 
-- Found HarfBuzz: /usr/lib/libharfbuzz.so (found version "2.6.1") 
-- Found CAIRO_EXTRAS: /usr/lib/libpangoft2-1.0.so  
```

And a random grep from the builddir:
```
[t@localhost build]$ grep -r /include/harfbuzz *
CMakeCache.txt:PKG_FONTCONFIG_CFLAGS:INTERNAL=-I/usr/include/freetype2;-I/usr/include/libpng16;-I/usr/include/harfbuzz;-I/usr/include/glib-2.0;-I/usr/lib/glib-2.0/include;-I/usr/include/uuid
CMakeCache.txt:PKG_FONTCONFIG_INCLUDE_DIRS:INTERNAL=/usr/include/freetype2;/usr/include/libpng16;/usr/include/harfbuzz;/usr/include/glib-2.0;/usr/lib/glib-2.0/include;/usr/include/uuid
CMakeCache.txt:PKG_FONTCONFIG_STATIC_CFLAGS:INTERNAL=-I/usr/include/freetype2;-I/usr/include/libpng16;-I/usr/include/harfbuzz;-I/usr/include/glib-2.0;-I/usr/lib/glib-2.0/include;-I/usr/include/uuid
CMakeCache.txt:PKG_FONTCONFIG_STATIC_INCLUDE_DIRS:INTERNAL=/usr/include/freetype2;/usr/include/libpng16;/usr/include/harfbuzz;/usr/include/glib-2.0;/usr/lib/glib-2.0/include;/usr/include/uuid
CMakeCache.txt:Pango_PKG_CFLAGS:INTERNAL=-I/usr/include/pango-1.0;-I/usr/include/glib-2.0;-I/usr/lib/glib-2.0/include;-I/usr/lib/libffi-3.2.1/include;-I/usr/include/harfbuzz;-I/usr/include/fribidi;-I/usr/include/freetype2;-I/usr/include/libpng16;-I/usr/include/uuid;-I/usr/include/cairo;-I/usr/include/pixman-1
CMakeCache.txt:Pango_PKG_INCLUDE_DIRS:INTERNAL=/usr/include/pango-1.0;/usr/include/glib-2.0;/usr/lib/glib-2.0/include;/usr/lib/libffi-3.2.1/include;/usr/include/harfbuzz;/usr/include/fribidi;/usr/include/freetype2;/usr/include/libpng16;/usr/include/uuid;/usr/include/cairo;/usr/include/pixman-1
CMakeCache.txt:Pango_PKG_STATIC_CFLAGS:INTERNAL=-I/usr/include/pango-1.0;-I/usr/include/glib-2.0;-I/usr/lib/glib-2.0/include;-I/usr/lib/libffi-3.2.1/include;-I/usr/include/harfbuzz;-I/usr/include/fribidi;-I/usr/include/freetype2;-I/usr/include/libpng16;-I/usr/include/uuid;-I/usr/include/cairo;-I/usr/include/pixman-1
CMakeCache.txt:Pango_PKG_STATIC_INCLUDE_DIRS:INTERNAL=/usr/include/pango-1.0;/usr/include/glib-2.0;/usr/lib/glib-2.0/include;/usr/lib/libffi-3.2.1/include;/usr/include/harfbuzz;/usr/include/fribidi;/usr/include/freetype2;/usr/include/libpng16;/usr/include/uuid;/usr/include/cairo;/usr/include/pixman-1
```

leamas, 2019-09-11:
Could you please submit the actual build error as well? Or, perhaps, attach a complete build log?

taruti, 2019-09-11:
The first message has the build error (with the patch it is identical). I can get you a clean build log and cmake files tomorrow if you want.

leamas, 2019-09-11:
Right, I *am* a sloppy reader. Sorry.

New attempt also patching libs/wxsvg/CMakelists.txt, which seems to be a somewhat better approach:

[0001-Handle-Pango-1.44-Harfbuz-dep-1443.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3602502/0001-Handle-Pango-1.44-Harfbuz-dep-1443.patch.gz)

If/when you have some time: could you test again?

leamas, 2019-09-16:
Nope, it doesn't work --- I have been spinning up a fedora 31 testbox with pango 1.44 and can reproduce this. Stay tuned...

taruti, 2019-09-17:
That PR fixes things for me, thank you for taking the time to fix things yet again!

leamas, 2019-09-21:
You're welcome.

If/when you have some spare cycles: I would really appreciate if you could make some basic testing of  #1457...

PR #1458 (2019-12-07): Handle Pango >=1.44 Harfbuz dep (Closes: #1453).
Recent pango libraries (1.44+) has a harfbuzz dependency.  Add a
new FindPango and FindHarfBuzz cmake finders and use them as
appropriate.

Tested on fedora 31 with pango-1.42.4-2

EDIT: For reference: closes #1453
commit 3e7f4ca9 (2019-12-07): Merge pull request #1458 from leamas/issue-1443

Handle Pango >=1.44 Harfbuz dep (Closes: #1453).
commit 15b85fe0 (2019-12-12): Merge pull request #1458 from leamas/issue-1443

Handle Pango >=1.44 Harfbuz dep (Closes: #1453).
closed 2019-12-07 by bdbcat