OpenCPN/OpenCPN #1454: Plugins not loading when compiling v5.0.0 from source

reneherrero, 2019-09-06:
The standard plugins included with OpenCPN are not loading correctly when compiling from source.  Am I missing something?

[image]
[image]
[image]
[image]
Some debug info:
* OS: Debian 10 Buster amd64
* System: Lenovo Thinkpad T480

reneherrero, 2019-09-06:
I should note that I tried a couple of things but they did **not** work:

1. Setting the LD_LIBRARY_PATH environment variable:
```
export LD_LIBRARY_PATH=/usr/local/lib/opencpn:$LD_LIBRARY_PATH
opencpn
```

2. Adding the library path to the system:
```
echo "/usr/local/lib/opencpn" | sudo tee -a /etc/ld.so.conf.d/opencpn.conf
sudo ldconfig
```

leamas, 2019-09-07:
Please look into the log file; here you will see from which directory your failing plugins are loaded.  My first guess is that you have some old plugins from older installations left around  which opencpn tries to load.

The plugins are not loaded by the runtime loader so LD_LIBRARY_PATH and friends are not involved. See the opencpn.1 manpage in the top source dir.

reneherrero, 2019-09-18:
That does not seem to be the issue.  I tried:
1. deleting all .so to make sure it's using the latest
2. another make install
3. delete log file
4. restart app

Same issue.  Here's the relevant piece in the logs:
```
03:38:50 CDT: PlugInManager: plugins loading from ~/.local/lib/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn
03:38:50 CDT: PlugInManager searching for PlugIns in location /home/rene/.local/lib/opencpn
03:38:50 CDT:    Directory /home/rene/.local/lib/opencpn does not exist.
03:38:50 CDT: PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
03:38:51 CDT:     Incompatible plugin detected: /usr/local/lib/opencpn/libwmm_pi.so
03:38:59 CDT:     Incompatible plugin detected: /usr/local/lib/opencpn/libdashboard_pi.so
03:39:01 CDT:     Incompatible plugin detected: /usr/local/lib/opencpn/libchartdldr_pi.so
03:39:03 CDT:     Incompatible plugin detected: /usr/local/lib/opencpn/libgrib_pi.so
```

leamas, 2019-09-18:
That's odd, like the say. Just to make one thing clear: there are many users out there building and running opencpn, so it's probably something with your installation..

What if you just remove the offending plugins in /usr/local/lib/opencpn before re-installing?

reneherrero, 2019-09-18:
Odd indeed, I've been always running v4.x from source.  This just started happening with the upgrade from Debian Stretch to Buster.

I tried a couple more things:
1. Tried building 4.8.8 -> same issue
2. Tried building 5.0.0 in Stretch in a VM -> plugins load correctly

Here's a thought: could this be a gcc issue?  Stretch uses v6 and Buster uses v8.  It might be a missing flag?

reneherrero, 2019-09-18:
Here's another fun fact: tried building v5.0.0 on an Ubuntu 18.04 machine -> also not loading the plugins.  That box uses gcc v7

leamas, 2019-09-19:
> could this be a gcc issue?

Perhaps, but probably not as simple as using v8 is broken. There are many users out there, and there would have been bugs in that case.

I am compiling upstream master on gcc 8.3.1 20190223 (Red Hat 8.3.1-2). I certainly do not recognize the problems you have.

reneherrero, 2019-09-20:
You where absolutely right, the problem was between the keyboard and the chair (me).

Debian seems to have made dependencies among packages a lot more granular in Buster.  I started over from scratch, looked very carefully at the `cmake ../` output and found that I was missing some libraries that cmake doesn't treat it as an error.  Symptoms manifest themselves at runtime.

Thanks for your help @leamas, much appreciated.  Closing this issue.

closed 2019-09-20 by reneherrero

leamas, 2019-09-21:
You're welcome :)