OpenCPN/OpenCPN #1459: wxWidgets 3.1.2 "special version" Source

scadaperspective, 2019-09-20:
Hi Dave and Folks,
Thanks for the efforts on the macOS version. 

Testing here so far looks a heap better. 

 To quote Pavel my crystal ball doesn't work so good so I am hoping someone has a zip / repo of the modified wxWidgets 3.1.2 library as I cannot seem to find any specific information on what exactly has been modified in the library to sort out the OpenGL issues.

looking at the travis batch file it points to a pre built automation library but not the source.  

All the best 

Ron Southworth

rgleason, 2019-09-20:
Ron, I don't know of any "special version" or "modified" wxWidgets 3.1.2.  Can you explain?

scadaperspective, 2019-09-21:
Hi Rick 

easiest way to highlight there are different library files used for main and the 5.0.1 mac version compiling using travis ....

Ron 

Sent from Ron’s iPhone while out and about!

rgleason, 2019-09-22:
Ron, thanks. Found this post by bdbcat
http://www.cruisersforum.com/forums/f134/opencpn-5-0-1-for-macos-mojave-10-14-5-a-220988-2.html#post2931593

scadaperspective, 2019-09-22:
Hi Rick

The two images were the easiest way to highlight there is a  different / special library being used. 

Glad Dave posted the version for people to try as it means he has not let go of the reigns completely.

I can say It looks like a promising “fix” to the opengl issue with macOS Mojave. 

Until the library is made available for the plugins to also be compiled that will make for a more complete test. 

At the moment it is promising not really conclusive....

All the best

Ron

Sent from Ron’s iPhone while out and about!

rgleason, 2019-09-22:
Ron, Dave, very much has the rein's  (not as in reign).
You two images attached here.
[image]
[image]

rgleason, 2019-09-22:
Ron I left a link for you in opencpn git issues Can you test squiddio version 1.0.8 for macos for me? It is in my repository under releases tab.Best Rick

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/OpenCPN/OpenCPN/issues/1459?email_source=notifications\u0026email_token=AASBENYHEIQWKJCRNFAM3EDQK3UAPA5CNFSM4IYSCEXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7I55PA#issuecomment-533847740",

"url": "https://github.com/OpenCPN/OpenCPN/issues/1459?email_source=notifications\u0026email_token=AASBENYHEIQWKJCRNFAM3EDQK3UAPA5CNFSM4IYSCEXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7I55PA#issuecomment-533847740",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]​

CarCode, 2019-09-23:
Sadly the developers of OpenCPN has gone their own way rather to regard what was going on at the wxWidgets people.
- The wxWidgets people has found a first patch to fix the problem with Mojave 10.14.5. However this patch does not find its way into the official wxWidgets trunk. Nevertheless the OpenCPN developers makes their own version 3.1.2 of wxWidgets using this patch.
- A short time later the wxWidgets people found a better solution to fix the problem. This solution is part of the wxWidgets version 3.1.3 now.
- See the discussions at: https://trac.wxwidgets.org/ticket/18402

So to compile OpenCPN for macOS use only wxWidgets version 3.1.3 and never any "special version"!
A warning to developers: Never publish a "special version" to avoid confusing people.

nohal, 2019-09-23:
There is no wxWidgets 3.1.3 yet. As you know.

CarCode, 2019-09-23:
Be patient. The official release will be in a few days (October).

nohal, 2019-09-23:
You are bitching about not using it in the past, so I am confused

rgleason, 2019-09-23:
Thanks for the "heads up" Carcode.

closed 2019-10-04 by scadaperspective