OpenCPN/OpenCPN #1461: Touch screen with no double click emu stuck on measure distance

mgrouch, 2019-09-25:
OpenCPN 5.0.0 becomes unusable
after I start ‘measure distance’ on
the touch screen with no double click emulation. It keeps adding points to the chart with no way exiting from this mode. 

A lot of touch 
screens are poorly supported on
raspberry pi and emulate only right click and mouse move.  

Maybe long mouse over could bring some pop up which if touched
could bring menu which opens on a double click. 

adding/removing waypoints and all other functionality is also unavailable without double click support. 

Thanks
MG

taruti, 2019-09-25:
Would https://github.com/JoseExposito/touchegg or https://github.com/iberianpig/fusuma help you?

mgrouch, 2019-09-25:
Possibly but they are not in available in Raspbian. 
Thanks

mgrouch, 2019-10-09:
Interestingly that Chromium web
browser on same pi with same touchscreen is able to zoom with 
two fingers touch and supports
right click with long touch.
Why same is not done in OpenCpn?
I have latest to date Raspbian version. 
Thanks

mgrouch, 2021-04-10:
there is workaround with evdev-rce

closed 2021-04-10 by mgrouch