OpenCPN/OpenCPN #1469: Can anyone tell me why Travis builds fail for osx and mingw?

transmitterdan, 2019-10-12:
Why does this build fail?

https://travis-ci.org/OpenCPN/OpenCPN/builds/596994775

rgleason, 2019-10-12:
Ok, I'll try.

From the logs
**Job 1563.3  osx**   
Line 1222  Do you have any ideas from this?
```
libmspub
==> Renamed Formulae
confluent-oss -> confluent-platform      transmission -> transmission-cli
gnatsd -> nats-server
==> Deleted Formulae

```

ENDING  looks like the file was built but it ended with a 1 for some reason.
```
Pruned 0 symbolic links and 6 directories from /usr/local
==> Downloading https://homebrew.bintray.com/bottles/libexif-0.6.21.mojave.bottl
==> Downloading from https://akamai.bintray.com/f2/f2825b7f043b7e1128a8a23413262
######################################################################## 100.0%
==> Pouring libexif-0.6.21.mojave.bottle.1.tar.gz
🍺  /usr/local/Cellar/libexif/0.6.21: 31 files, 785KB
_**The command "./ci/travis-build-osx.sh" exited with 1.**_
Done. Your build exited with 1.
```

**Job 1563.4   mingw**
The system installed ok, but there don't appear to be any builds.  --could it be a programming problem?

```
Complete!
+ sudo dnf copr enable -y --nogpgcheck leamas/opencpn-mingw
Error: This repository does not have any builds yet so you cannot enable it now.
The command "./ci/travis-build-mingw.sh" exited with 1.
```

rgleason, 2019-10-12:
TDan, maybe the travis file's  osx build needs to be updated?

leamas, 2019-10-12:
This is really two problems:

The first is about some new packages available in the osx image -- homebrew generates an error when trying to install an already installed package. Fixes available in #1432 and #1456 

The second should not happen when rebuilding master (?). That said, it's about fedora dropping support for 28, moving to 29 (at least). A simple s/28/29/ makes it, but TBH I don't know in which file in your case.

leamas, 2019-10-12:
Expanding on above: s/28/29/ in ci/travis-build-mingw.sh should fix it. Guess I need to write a PR...

leamas, 2019-10-12:
Fix for the mingw builder (patch): 

[0001-ci-mingw-Update-fedora-builders-to-supported-f29.patch.gz](https://github.com/OpenCPN/OpenCPN/files/3721062/0001-ci-mingw-Update-fedora-builders-to-supported-f29.patch.gz)

transmitterdan, 2019-10-12:
@leamas thanks for the help.  I think those patches have corrected the builds for my PRs.

closed 2019-10-12 by transmitterdan

douwefokkema, 2019-10-13:
I closed my PR and resubmitted: https://github.com/OpenCPN/OpenCPN/pull/1470
Still going wrong.

rgleason, 2019-10-13:
leamas osx fix  https://github.com/OpenCPN/OpenCPN/pull/1456
leamas mingw fix  see the attached patch https://github.com/OpenCPN/OpenCPN/issues/1432#issue-462335823