OpenCPN/OpenCPN #1477: New plugin freezes OpenCPN's settings window

JohannCahier, 2019-10-25:
Hello,

I'm pretty new to OpenCPN.
I'm using it on x86_64, arch linux. I used the recipe found on the AUR (https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=opencpn)

On the other hand I have cloned : https://github.com/msis/ocpn_pi_tutorial which claims to be a raw base to start an OCPN plugin development.
I compiled it using
`mkdir build && cd build && cmake .. && make`
Then copied the produced .so file into `/usr/lib/opencpn`

(I also observed the behaviour described below, with a down-stripped fork of DR_pi... So basically an empty plugin).

Finally OpenCPN is restarted : everything goes fine as long as I do not enter in settings->extensions menu.

- Good points : 
The plugin is well-listed in the "extension" window. 
Graphical elements (buttons, etc) in this window are reactive to mouse clic (focus goes to the last clicked button)...

- Bad points:
BUT the window itself never closes !
Neither the Ok/Cancel/Apply button inside the window, not the window decoration buttons are effective.

Do someone have more information on this ?
Did the 5.0.0 broke

Hakansv, 2019-10-25:
Read this:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux
and follow all steps rigorously. Not less sudo make install.

JohannCahier, 2019-10-28:
Hello, thanks for the suggestion. It indeed actually worked. 
I just had to specify harfbuzz include directory on the command line :
`CXXFLAGS=-I/usr/include/harfbuzz cmake ..`
With this option, I built seamlessly.

And, I have my plugin traces on the console, so 

Do you have any clue why the v5.0.0 I built and installed formerly exhibit such a behavior ?

Thanks again !

rgleason, 2021-04-14:
Ok Johann,  Can you please close this Issue now? Thanks.

bdbcat, 2021-04-29:
Closing, inactive

closed 2021-04-29 by bdbcat