OpenCPN/OpenCPN #1478: Handle Navico's invalid DPT sentence

caesar, 2019-10-26:
Navico devices (at least all I have tested – several current and recent past Simrad and B&G plotters, and a couple of older Lowrance fishfinders) emit the keel offset in their output `DPT` sentence in an invalid format with a second minus sign after the decimal place, eg `-1.-4` instead of `-1.4`.

OpenCPN currently parses this as meaning `-1` – ie, it truncates the number after the decimal place. This results in depth soundings **up to 0.9m deeper than the true depth**. (This is only an issue with negative offsets – to the keel – and so the error is always positive.)

Clearly, the bug is on Navico's side. But since it's present in hundreds of thousands (millions?) of shipped devices, we would do well to handle it gracefully. Ideally by working around it and interpreting `-1.-4` as `-1.4`, but failing that, by dropping the invalid data altogether.
The latter option would be far less satisfactory, but *much* better than showing deeper water than the true depth.

Example from a new Simrad NSS12 evo2 plotter (keel offset is -1.1m):
`$SDDPT,24.6,-1.-1,*4B`

transmitterdan, 2019-10-28:
The DPT message is only used by Dashboard.  The parser can be easily modified to ignore extra - characters in the offset field. For example:

```diff --git a/plugins/dashboard_pi/src/nmea0183/dpt.cpp b/plugins/dashboard_pi/src/nmea0183/dpt.cpp 
index c42d83cba..05e0ce535 100644
--- a/plugins/dashboard_pi/src/nmea0183/dpt.cpp
+++ b/plugins/dashboard_pi/src/nmea0183/dpt.cpp
@@ -106,8 +106,14 @@ bool DPT::Parse( const SENTENCE& sentence )
}

    DepthMeters                = sentence.Double( 1 );
-   OffsetFromTransducerMeters = sentence.Double( 2 );
+   wxString offset = sentence.Field( 2 );
+   int minus = offset.find_last_of('-');
+   if (minus != wxNOT_FOUND && minus > 0)
+   {   // Remove extra - from transducer offset
+       wxString first = offset.substr(0, minus);
+       offset = first + offset.Mid(minus + 1);
+   }
+   offset.ToDouble(&OffsetFromTransducerMeters);
    return( TRUE );
 }
```

But is this something that should be done only for this one field in this one sentence?  Are there other sentences where Navico systems misbehave in this way?

bcn2, 2019-10-28:
Did you contact Navico and if so, what has been the reply?
I would call it a scandal if they would maintain a bug deliberately.

caesar, 2019-10-28:
> But is this something that should be done only for this one field in this one sentence? Are there other sentences where Navico systems misbehave in this way?

I haven't seen any others, but I'm not sure what other fields might contain a negative float. I have some recent NMEA logs from some Navico devices so I'll investigate further.

> Did you contact Navico and if so, what has been the reply?

I haven't (I'm only just back online after a few months without real internet access) but I will.
That said, we need to deal with this anyway, because even if Navico fix it, it's already present in millions of shipped devices.

caesar, 2019-10-28:
Ok… so, I can confirm that this error happens in other places, but *not in all* negative floats.

For example, see the following XDR sentence, where the negative angle of heel is invalid (`-1.-2`) but the negative trim angle is valid (`-6.6`).
As for the rudder angle, it's pretty weird (`-25.0612.0`), I can't work out what's going on there nor why they'd even try and give rudder angle to 5 decimal places of a degree…
```
$IIXDR,C,4.4,C,AIRTEMP,A,-1.-2,D,HEEL,A,8.1,D,TRIM,P,0.989,B,BARO,A,-25.0612.0,D,RUDDER*1A
```
But also note this one – when the negative number is less than `-1`, the leading minus sign is dropped! So we have `0.-5` instead of `-0.5`.
```
$IIXDR,C,3.8,C,AIRTEMP,A,0.-5,D,HEEL,A,-6.6,D,TRIM,P,0.989,B,BARO,A,0.939651.9,D,RUDDER*09
```

I haven't found any sentences where this happens other than DPT and XDR.

caesar, 2019-11-14:
I contacted Navico the day after opening this issue, and have just heard back from them.
They say they are able to reproduce the bug and it will be fixed in a future state update; possibly the next one.

Regardless, we should handle this gracefully, as the bug is present in millions of shipped devices and 99% will never be updated.

transmitterdan, 2019-11-14:
I have tested a fix for the bad DPT sentence. Will submit a PR soon.

transmitterdan, 2019-11-16:
See PR: https://github.com/OpenCPN/OpenCPN/pull/1486

rgleason, 2021-04-14:
I believe this has been completed and should be closed https://github.com/OpenCPN/OpenCPN/pull/1486
but I could not find the cleaner PR.

transmitterdan, 2021-04-17:
A PR for the DPT case was merged into master.  However, I never went back and made the fix universal for all floats that might have extra - symbol in a float string.  I will submit a PR that handles malformed message for all float strings in an NMEA message.

caesar, 2021-04-17:
Also, @transmitterdan, your PR #1518 only fixes the display in the Dashboard plugin.
Wouldn't a better approach be to fix the _input_ when OpenCPN reads the NMEA data, so that the fixed data is available throughout the app (eg for other plugins too)?

(Also, sorry not to have followed this up with a PR myself! I will see if I can make time at some point but it probably won't be for at least a few weeks I'm afraid.)

caesar, 2021-04-17:
Looking at this again @transmitterdan (and sorry not to have noticed before your PR #1518 was merged…) I think your approach in that patch for `DPT` will only work in the subset of cases where the keel offset is greater than 1m (in other words `offset < -1.0`).
For cases where `0.0 < offset < 1.0`, the patch will result in a positive output when it should have been negative (eg `0.-5` will result in an output of `0.5`).

See [my comment above](https://github.com/OpenCPN/OpenCPN/issues/1478#issuecomment-547187123):
> when the negative number is less than `-1`, the leading minus sign is dropped! So we have `0.-5` instead of `-0.5`

So an approach may be necessary which specifically looks for that second invalid minus sign, and if it is present, ensures the resulting offset is negative *even if the integer part is zero*.

transmitterdan, 2021-04-17:
Ok, I'll look at the patch again.

transmitterdan, 2021-04-17:
OpenCPN does not parse the NMEA message. It just forwards it. Fixing the problem in OpenCPN is much more complex than it seems. One has to find the error, fix it, and then re-encode the message with a new checksum because the original checksum will be invalid. 

I'll look at how disruptive that might be but do not get hopeful for a universal fix.

caesar, 2021-04-17:
Good point, I hadn't thought about the checksum handling – though it might still be worthwhile as it would be potentially confusing to have different values displayed in different places… depending on the trade-offs involved. (These are just random thoughts on my part, as I'm not familiar with that part of OpenCPN's architecture.)

An alternative to recalculating the checksum might just be to drop it; a valid sentence doesn't require a checksum and there's (practically speaking) no risk of corruption once the sentence is in memory within OCPN.
Unless there are parts of OCPN that do require that a sentence has a checksum… more trade-offs to consider I guess.

caesar, 2021-04-17:
I should add that I can also see a perfectly valid argument for *not* messing centrally with the NMEA sentences received, as (for example) the VDR plugin would then presumably end up logging the modified sentences instead of the raw ones, which could be undesirable.

transmitterdan, 2021-04-18:
If other plugins want to emulate DashBoard they can steal the NMEA decoder code from DashBoard.

rgleason, 2021-04-19:
Caesar do you think this should be close now?  You and the Moderator are the only ones who can do it. 
Or is there something else that could be done that does not mess up the nmea stream?
Perhaps you could make a special Plugin that will display the correction? That would be self contained.
Call it FixNavicoDPT maybe?

Anyway can you close this now?

caesar, 2021-04-19:
Rick, I think it shouldn't be closed yet as the current fix actually makes
the problem worse (doubles the error) in a subset of cases. I think Dan was
planning to look at that again.

transmitterdan, 2021-04-19:
@ceasar, I don't understand what you mean by "doubling the error".

caesar, 2021-04-19:
Sorry I wasn't clear @transmitterdan. I was referring to the situation I mentioned in [my previous comment](https://github.com/OpenCPN/OpenCPN/issues/1478#issuecomment-821807415) where:
> For cases where 0.0 > offset > -1.0, the patch will result in a positive output when it should have been negative (eg 0.-5 will result in an output of 0.5). Potentially this could mean the displayed water depth is up to 1.8m deeper than the true depth (in the case of an offset of -0.9 which will be encoded in Navico's DPT sentence as 0.-9).

To clarify, the original situation (before #1518) was that Navico's invalid float values (eg `-1.-5` or `0.-9`) caused the keel offset value to be *truncated*, causing an over-reading equal to the decimal part of the offset (ie, an over-reading of up to 0.9m).

The current situation (after #1518) is that
- *where the keel offset is 1m or more* (offset ≤ -1.0) the displayed depth will be correct (because your fix converts eg `-1.-5` to `-1.5`)
- ***but***, *where the keel offset is less than 1m* (-1.0 < offset < 0.0) the displayed depth will over-read *by double the decimal part of the offset* (because your patch will convert eg `0.-9` to `0.9` when it should be `-0.9` – resulting in an over-reading of 1.8m in this example).

It was the latter case I was referring to when I mentioned "doubling the error".

My suggested fix would be
> So an approach may be necessary which specifically looks for that second invalid minus sign, and if it is present, ensures the resulting offset is negative even if the integer part is zero.

I am happy to look at this at some point but I won't have time in the short term.

Either way at the moment the issue is not resolved @rgleason 🙂

caesar, 2021-04-19:
Additionally, we discussed generalising the patch (even if just within the Dashboard plugin?) to cover other sentences such as `XDR` – I am not sure if you are still interested in doing this?

I could retitle the issue to cover the general case (not just `DPT`) and leave open until that is solved, unless maybe Dave @bdbcat reckons working around issues caused by external bugs is feature creep…?

transmitterdan, 2021-04-19:
@caesar,  I think if you check the latest PR it does indeed handle the general case.  If any float value contains misplaced '-' symbols it will be considered as a negative number.  It is done for all sentences and fields that contain floating point values.

rgleason, 2021-04-19:
Caesar, don't forget to close this if TDan has solved it for you. Thanks.

caesar, 2021-04-20:
Apologies @transmitterdan, I wasn't aware that you had submitted a second PR #2212 (it looks like you forgot to tag it with `fixes #1478`, which would have linked it here and also automatically closed this issue when it was merged).

Your PR #2212 looks like it should solve the issue fully for the Dashboard (I haven't tested though). Cheers! 👍 

I will close this issue for now, but will re-open if I identify any further areas where this Navico bug needs to be worked around.

closed 2021-04-20 by caesar