OpenCPN/OpenCPN #1479: Small chart display on fresh new compilation (viewport issue on mac?)

theeko74, 2019-11-07:
I just compiled OpenCPN from scratch today, and it appears that I have an issue with the display (as enclosed). I deleted every Preferences files I have but the problem is still there. Would it be an issue with the viewport ? 

[image]

Extracts from the log
wxWidgets 3.1.2 Apple Mac OS X 64 bit wxMac
Detected display size (horizontal): 601 mm
Metrics:  m_display_size_mm: 601     wxDisplaySize:  3360:1890
(It does not change the problem if I change any of these parameters).

Any idea?

rgleason, 2019-11-07:
Hi Theeko, it may be helpful if you post a link to this from the Cruiser Forum beta test/ technical thread here.  http://www.cruisersforum.com/forums/f134/beta-test-technical-30929.html

CarCode, 2019-11-09:
See here:
https://trac.wxwidgets.org/ticket/18402

rgleason, 2019-11-09:
Thank you Carcode.
"Can be reproduced on any macOS 10.14.5 with sample builded with 10.14 SDK. "
 "(10.14.4 work ok)"
Fixed 3 months ago, but would require a newer version of wxWidgets.
Possibly use 10.14.4 instead.

CarCode, 2019-11-09:
Close.
Was fixed with OpenCPN v5.0.1 but sadly not in the right way.

rgleason, 2019-11-09:
Carcode that's right, I had forgotten about that fix.  Theeko, that's the version you need.
From Web downloads page 
https://opencpn.org/OpenCPN/info/downloadopencpn.html
Mac
OpenCPN 5.0.1 for Mac OS X 10.7+
SHA256:
b3da4a758d898a9a2f79ba2618ff6cf9529b06dd1ccba14f4dd3e4996660b143

This should be closed now. Theeko would you please close it?

theeko74, 2019-11-14:
So which latest version of wxWidgets should I use (and still compatible with OpenCPN) ?

rgleason, 2019-11-14:
Theeko, the version I believe that you need for Opencpn 5.0 is wxWidgets 3.1.2

See the Dev Wiki, Compiling for MacOS
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_mac_osx

rgleason, 2019-11-15:
Theeko,  good it compiled.  Now try looking for Github repository/TAG  for the MacOS 5.01 version of Opencpn because there were some MacOS problems that Bdbcat sorted out, perhaps that will help!

PS I can't find version 5.01 on github.  Sorry. Maybe someone else knows more.

rgleason, 2021-04-14:
I believe this can be closed now.  Theeko 74 will do please close it?

nohal, 2021-04-24:
This exact problem will be back with future wxWidgets (https://github.com/OpenCPN/OpenCPN/issues/2131) due to their changed behavior.

bdbcat, 2021-04-30:
Closing, duplicate.

closed 2021-04-30 by bdbcat