OpenCPN/OpenCPN #1480: Documentation alternative - so it can be packaged

rgleason, 2019-11-10:
Please refer to this beginning effort for OpenPlotter
https://openplotter.readthedocs.io/en/latest/dashboards/influxdb_grafana.html

1. All OpenCPN Documentation needs to be updated to v5.0.0.
-New screenshots, organized by a naming convention.
-Integrate the new features into the documentation.
-Cleanup and simplify.
2. Alec Leamas asked for the ability to package our Wiki Documentation, advising it needs to be opensource.  We found that the best way to achieve that from Docuwiki was to create searchable EPUB files. This is quite successful and relatively easy to do, but it cannot be done frequently or on demand.
3. Typical difficulties and inability to have documentation ready at the time of releases.
4. The OpenPlotter Documentation may serve as an example for us to consider, it would mean a lot of work changing to another new system however.  Some examples and links for consideration. It is a work in progress and largely empty. It is[ editable via Git](https://github.com/openplotter/docs/blob/master/description/what_is_openplotter.rst) as I understand it.

[What is Openplotter](https://openplotter.readthedocs.io/en/latest/description/what_is_openplotter.html)
[Dashboard Influxdb](https://openplotter.readthedocs.io/en/latest/dashboards/influxdb_grafana.html) 
[Dashboard NodeRed](https://openplotter.readthedocs.io/en/latest/dashboards/nodered_dashboard.html)

Thoughts?

leamas, 2019-11-27:
Changing documentation format is a huge task not to be taken lightly. However, if doing this the new system should certainly be based on an open process translating sources to docs. This will support that users could file pull requests also against the docs which would simplify things a lot. I suppose this is what you refer to as *editable via Git*, right?

In the developer community the dominating tool are simple wikis. They are light-weight, easy to edit and of course using text-based sources normally handled by a VCS like git. I intend to explore the possibility to use the wiki built in into github to document the plugin installer -- perhaps we could make some conclusions from that later.

leamas, 2019-11-27:
I need to be more clear: I don't think a wiki is a viable way to provide end-user docs. But, it could perhaps be useful for developer info.

rgleason, 2020-01-06:
Alec, I agree. You are using the github wiki for Plugin Manager Documentation and it looks userful.
https://github.com/leamas/OpenCPN/wiki  It handles images well too.
It is a huge task, but we will need to rewrite it anyway at some point soon.

I think the User Documentation should be split up into smaller packages starting with  
**Getting Started + Toolbar + Basic + Default Plugins**  
First one to work on and include in software.

Then  Advanced Charts, Plugins, etc. etc
Closing this now.

closed 2020-01-06 by rgleason